ARM-software/CMSIS_5 #1421: rt_init_stack does not initialize FPU stack

CurtisAMU, 2022-02-14:
rt_init_stack in CMSIS/RTOS/RTX/SRC/HAL_CM.c initializes the stack of a new thread - including main(). If the FPU is enabled, the run-time system sets FPSCR up according to the chosen FP model, for instance, --fpmode=std with armcc, which will set DN and FZ in FPSCR. Interestingly, the run-time system does not initialize FPDSCR, though. FPSCR is part of the stack frame saved if the FPU is enabled and lazy or automatic context saving is configured. However, rt_init_stack does not handle the FPU stack.
With the current implementation of CMSIS-RTOS1, DN and FZ are **cleared** when entring main() or any other thread the first time, leaving the FPU in a state that is different from what fpmode may assume. To avoid surprises, the initialization of a thread should inherit the current FP configuration. A thread may alter it later for itself.

RobertRostohar, 2022-02-14:
Please note that RTX4 (implements CMSIS-RTOS1 API) has been superseded by RTX5 (implements CMSIS-RTOS2 API) a long time ago. It is strongly recommended to upgrade to RTX5.

Every thread is created without floating-point context. This reduces the overhead of stacking FPU registers if floating point is not used. Therefore rt_init_stack (called at thread creation) does not handle extended stack frame with FPU registers but just the basic stack frame.

When floating point is used in a thread the floating-point context becomes active and certain bits (AHP, DN, FZ, RMode) in FPSCR are initialized from FPDSCR.

It is expected that FPSCR and FPDSCR are properly initialized before floating point is being used (typically before threads are created). FPSCR is initialized by the CLib run-time however it seems that FPDSCR is not. A simple solution would be to copy the value from FPSCR (after initialized by CLib run-time) to FPDSCR before starting to use floating point operations.

CurtisAMU, 2022-02-14:
Our product is in production, and we cannot change the OS now. RTOS2 will be used for new projects.

It might be worthwhile to document the OS's assumptions about the use of FP (same for RTOS2 I suppose).
It seems odd to me that ARM-Keil's _fp_init() initializes FPSCR but not FPDSCR. Unfortunately, in the ARM compiler v5 environment, there is no hook for user code between _fp_init and entering main. Of course, one can call _fp_init() a second time in main and then copy the configuration over to FPDSCR.

CurtisAMU, 2022-02-14:
ARM will not change the behavior of _fp_init(), and setting up FPDSCR in main is cumbersome as main usually does not run in privileged mode.
The minimum would be to document the caveats. When creating a new thread, it is not clear what FP configuration it will have. If an application modified FPDSCR for use in an exception, a thread created later might not receive the configuration that matches ARM-Keil's fpmode that will be active when the thread is running. Calling _fp_init for each thread would make the threads independent from FPDSCR as their specific FPSCR values are saved on their respective stacks.

RobertRostohar, 2022-02-15:
> It seems odd to me that ARM-Keil's _fp_init() initializes FPSCR but not FPDSCR.

Unfortunately this is the default behavior of Arm CLib which only cares about a single floating point context.
However this can be easily fixed by using the Super/Sub concept and extend _fp_init() as following:

```
extern void $Super$$_fp_init (void);

void $Sub$$_fp_init (void) {
  $Super$$_fp_init();
  FPU->FPDSCR = __get_FPSCR();
}
```

This should properly setup also FPDSCR and no other provision should be needed.

> It might be worthwhile to document the OS's assumptions about the use of FP 

Yes, this and the above solution should be documented.

IngmarPaetzold, 2022-09-27:
Hi, I just stumbled over this issue while scanning RTX 5.5.4 release notes. Our project currently uses RTX 5.5.3, and uses a few threads and OS timers, which work with floating point values here and there.
These are just generated with osThreadNew() and osTimerNew().
Are there any potential implications to expect? Does floating point usage need to be activated or explicitly initialized, and we were not aware of?

RobertRostohar, 2022-09-28:
Proper FPU initialization should be added as discussed in this thread. This has been added in RTX 5.5.4. When using older RTX 5.5.3 the FPDSCR initialization should be added. See code above or take a look at the commit (28d54ba).

IngmarPaetzold, 2022-09-28:
I manually inserted the fix into **rtx_lib.c**, but it was excluded from compiling because **__FPU_PRESENT** is _not_ 1.

This definition would be available via **rtx_lib.h** (includes **rtx_core_c.h**, includes **RTE_Components.h**, eventually includes the actual device definition (here **stm32f417xx.h**)).
But none of these are seen by **rtx_lib.c**, despite the similar name.
In my setup, **rtx_lib.h** is only included in RTX subsystem source files such as **rtx_timer.c, rtx_thread.c, rtc_kernel.c** etc.).
Somehow it seems that **rtx_lib.c** is missing its "own" header file, isn't it?

(My setup: STM32F417, Keil µVision toolchain, RTX 5.5.3)

JonatanAntoni, 2022-09-28:
Yes, indeed. Please carefully inspect the fix:
https://github.com/ARM-software/CMSIS_5/commit/28d54ba19eddb868340dfb86774d77e8dfa65e72#diff-072af3a2ee4aa81701219783203b383260a6be87edd833dd59d558b07482b355R29-R32

IngmarPaetzold, 2022-09-28:
O yes, my apologies... I was kind of sleeping... Thanks.