ARM-software/CMSIS_5 #1424: CMISS-DSP:compile fft bin failed

listzhp, 2022-02-23:
1.I under the path "cmsis/CMSIS/DSP/Examples/ARM/arm_fft_bin_example",2.and mkdir build,3.then cd build;4.cmake -DCMAKE_PREFIX_PATH="gcc-arm-none-eabi-10.3-2021.07-x86_64-linux/gcc-arm-none-eabi-10.3-2021.07/bin/"     -DCMAKE_TOOLCHAIN_FILE="../../../../gcc.cmake"     -DARM_CPU="cortex-m55"     -DROOT="../../../../../.."   -DPLATFORM="FVP" -G "Unix Makefiles" ..;5.there generate bin_dsp success,but would output error: undefined reference to `rearranged_twiddle_tab_stride1_arr_4096_f32'.......arm_cfft_init_f32.c:(.text.arm_cfft_radix4by2_rearrange_twiddles_f32+0x188): undefined reference to `rearranged_twiddle_tab_stride3_arr_16_f32' and so on. what I need fix? thanks. I just want use rfft

JonatanAntoni, 2022-02-23:
@listzhp,

May I ask you to reformat your request, please? It is very hard to read. You could try to use Markdown to format lists and code.

Thanks,
Jonatan

listzhp, 2022-02-23:
> @listzhp,
> 
> May I ask you to reformat your request, please? It is very hard to read. You could try to use Markdown to format lists and code.
> 
> Thanks, Jonatan

Ok, I try to fix, thanks!

christophe0606, 2022-02-23:
@listzhp Hi. You need to add the option 

`-DHELIUM=ON`

I did not want to have a complex logic in the cmake to parse the ARM_CPU and deduce if Helium is used or not. So it has to be specified separately.

Also, depending on your gcc version, you may also need to add 

`-DCMAKE_C_FLAGS="-flax-vector-conversions"`

if you get some warning or errors about vector conversions.

listzhp, 2022-02-24:
> DHELIUM
@christophe0606 Thanks your reply. If the cpu is cortex-m55, "-DHELIUM=ON" is supported , right? I'not sure about that.
Thanks again.

christophe0606, 2022-02-24:
@listzhp Indeed, if you use "cortex-m55" then Helium is implied. 
Otherwise you would need to use something like "cortex-m55+nomve.fp+fp.dp"

listzhp, 2022-02-24:
@christophe0606 .OK. If I want get best performance on m55, what's your suggestion about the off-on. Thank you so much!
...
option(NEON "Neon acceleration" ON)
option(NEONEXPERIMENTAL "Neon experimental acceleration" OFF)
option(HELIUMEXPERIMENTAL "Helium experimental acceleration" OFF)
option(LOOPUNROLL "Loop unrolling" ON)
option(ROUNDING "Rounding" OFF)
option(MATRIXCHECK "Matrix Checks" OFF)
option(HELIUM "Helium acceleration (MVEF and MVEI supported)" ON)
option(MVEF "MVEF intrinsics supported" OFF)
option(MVEI "MVEI intrinsics supported" OFF)
option(MVEFLOAT16 "Float16 MVE intrinsics supported" OFF)
option(DISABLEFLOAT16 "Disable building float16 kernels" OFF)
option(HOST "Build for host" OFF)
...

christophe0606, 2022-02-24:
@listzhp You add those settings:

`-DCMAKE_C_FLAGS="-Ofast -ffast-math"`

Note that clang may be better than gcc currently for performance on M55.

listzhp, 2022-02-28:
> 
hi,another question. If the input data length is 512, I should set arm_rfft_fast_init_f32(&fftInstance, 256) or arm_rfft_fast_init_f32(&fftInstance, 512)? where the second param is the output length. right?

listzhp, 2022-02-28:
> > 
> 
> hi,another confusion. If the input data length is 512, I should set arm_rfft_fast_init_f32(&fftInstance, 256) or arm_rfft_fast_init_f32(&fftInstance, 512)? where the second param is the output length. right?

christophe0606, 2022-02-28:
@listzhp It is the number of real samples.
So you should use 512.
The output will consist of 512 samples.

If the mathematical output is y, then cmsis-dsp output is: 

[y(0),**Re(y(n/2))**,Re(y(1)),Im(y(1)),...,)

whereas the python scipy.fftpack.rfft output is

[y(0),Re(y(1)),Im(y(1)),...,**Re(y(n/2))**]

and n= 512

closed 2022-03-09 by christophe0606