ARM-software/CMSIS_5 #1429: How to get best performance of CMSIS-NN

listzhp, 2022-03-02:
Hi, I want use cmsis-nn in tflite-micro, and the cpu is m55, I just set OPTIMIZED_KERNEL_DIR=cmsis_nn, is that enough? Do I need set the macros  manually? for example:
ARM_MATH_DSP:
Define macro ARM_MATH_DSP, If the silicon supports DSP instructions(DSP extension).

ARM_MATH_MVEI:
Define macro ARM_MATH_MVEI, If the silicon supports M-Profile Vector Extension.

ARM_MATH_AUTOVECTORIZE Used in conjucture with ARM_MATH_MVEI to let the compiler auto vectorize for the functions that uses inline assembly. It does not affect functions that use C or intrinsics.

ARM_NN_TRUNCATE:
Define macro ARM_NN_TRUNCATE to use floor instead of round-to-the-nearest-int for the computation.

Thanks.

mansnils, 2022-03-02:
Hi @listzhp
Yes that is enough. Note that it will use the downloaded CMSIS. If you want to specify your own location you could add CMSIS_PATH=<external/path/to/cmsis/>. Here are some examples: https://github.com/tensorflow/tflite-micro/blob/main/tensorflow/lite/micro/kernels/cmsis_nn/README.md
ARM_MATH_DSP and ARM_MATH_MVEI will be automatically defined by the compiler, providing the compiler supports it. If you use the above examples the downloaded compiler will support it.

felix-johnny, 2022-03-14:
@listzhp I believe the question is answered and I'll close the ticket. But, if there is a follow up or any new question, just ping us and we are happy to help!

closed 2022-03-14 by felix-johnny