ARM-software/CMSIS_5 #1430: CMSIS fft and kiss-fft get different results

listzhp, 2022-03-04:
Hi,I use cmsis arm_rfft_fast_f32() and kiss fft,but get different results, I don't know why.
Firstly, I set input data length 512, and input_data[i]=1.3i;(0.0 1.3 2.6 3.9 5.2........)
Then, 
state = kiss_fftr_alloc(512, 0, 0, 0);
kiss_fftr(state, input_data, out);
kiss-fft out:170060 0 -332 54237 -332 27117 -332 18077 -332 13556.......
and then,
arm_rfft_fast_init_f32(s,512);
arm_rfft_fast_f32(s, input_data, out, 0);
cmiss-fft out:84864 85196 -27450 26784 -13889 13223 -9368 8702 -7107 6441.......

so I am not sure I use cmsis fft correctly, would you have some advice? Thanks!

christophe0606, 2022-03-04:
I have tested with the CMSIS-DSP Python wrapper:
`pip install cmsisdsp` 

(It is a source package so a compiler is needed)

Here is my Python code
```python
import numpy as np
import cmsisdsp as dsp

nb = 512
signal = np.array(list(range(nb))) * 1.3

rfftf32=dsp.arm_rfft_fast_instance_f32()
status=dsp.arm_rfft_fast_init_f32(rfftf32,nb)
result = dsp.arm_rfft_fast_f32(rfftf32,signal,0)
print(result[0:10])
```

I am getting:

`[170060.8       -332.79688   -332.79962  54237.285     -332.80014
  27117.621     -332.80008  18077.281     -332.8001   13556.77   ]`

The second sample is different and it is normal since the CMSIS-DSP RFFT is packing the real value at FFT[n] inside FFT[0] (if FFT is seen as an array of complex).

But otherwise it is the same as kissfft.

So I have no idea what is your problem.

You may check the status returned by `arm_rfft_fast_init_f32` and check that `input_data` can be modified (since it is also modified by the CMSIS-DSP RFFT).

closed 2022-03-14 by christophe0606

erhnam, 2022-07-13:
You are using the input_data list after using it with kiss_fft, which modifies that input list.