ARM-software/CMSIS_5 #1431: RTX5 SysTick Handler breaking change

BrunoCoppi, 2022-03-09:
I would like to notify you that the change in osRtxTick_Handler introduces an issue that shows up if the SysTick Handler is called before the osKernelInitialize function is called.
The "thread" local variable is used without a "not-null" check as it was in CMSIS v5.7.0.
Initializing the SysTick before RTX is obviuosly an error, but with STM32 devices, STM32CubeMX generates code that calls the HAL_Init function which, in turn, calls the HAL_InitTick function that initializes the SysTick enabling the interrupt.
A possible solution is to override the HAL_InitTick (which is marked __weak), but the STM32 DFP documentation has to be updated: reimplementing the HAL_GetTick function is no more sufficient.

GuentherMartin, 2022-03-10:
Hello @BrunoCoppi, Thank you for pointing out.
STM32 DFP documentation will be updated.

I suggest to add following code to the STM32CubeMX generated main.c file:
```
/* USER CODE BEGIN 0 */
/**
  * Override default HAL_GetTick function
  */
uint32_t HAL_GetTick (void) {
  static uint32_t ticks = 0U;
         uint32_t i;

  if (osKernelGetState () == osKernelRunning) {
    return ((uint32_t)osKernelGetTickCount());
  }

  /* If Kernel is not running wait approximately 1 ms then increment 
     and return auxiliary tick counter value */
  for (i = (SystemCoreClock >> 14U); i > 0U; i--) {
    __NOP(); __NOP(); __NOP(); __NOP(); __NOP(); __NOP();
    __NOP(); __NOP(); __NOP(); __NOP(); __NOP(); __NOP();
  }
  return ++ticks;
}

/**
  * Override default HAL_InitTick function
  */
HAL_StatusTypeDef HAL_InitTick(uint32_t TickPriority) {
  
  UNUSED(TickPriority);

  return HAL_OK;
}
/* USER CODE END 0 */
```

closed 2022-03-10 by GuentherMartin

RobertRostohar, 2022-03-10:
This is a known issues and was already discussed in #1330.

As pointed out the HAL_InitTick function needs to be updated as shown.