ARM-software/CMSIS_5 #1440: [CoreValidation] How to build?

liyuankang, 2022-03-17:
I tried to run the CoreValidation/Tests/build.py, but failed:
```
Traceback (most recent call last):
  File "builder.py", line 8, in <module>
    from buildutils.builder import Device, Compiler, Axis, Step, BuildStep, RunM
odelStep, Builder, Filter
ImportError: No module named builder
```

@JonatanAntoni   @GuentherMartin 
Is there README to introduce the usage of CoreValidation? Thanks.

JonatanAntoni, 2022-03-17:
Hi @liyuankang,

The `builder.py` is deprecated and not maintained anymore and should be removed. The replacement if `build.py` which uses `python-matrix-runner` module. But to build and run the CoreValidation tests you'd need another unpublished Python module which is deprecated as well and won't become available.

The official way for building projects in CI is denoted to be CMSIS-Build. Once this tool is powerfull and stable enough we need to rework the CoreValidation tests to use it. By then you should be able to build and run the tests given you have the required tools. 

The test source itself should work but for the time being you'd need to create working projects yourself depending on your tool environment. Executing the tests would need some sort of emulator as the tests are hardware-specific and cannot be run on host computers.

Cheers,
Jonatan

liyuankang, 2022-03-18:
Hi @JonatanAntoni,

Thanks for your quickly and detailed reply!
I have created a MDK project for CM3. It can pass the build. I will try to run and test it.
CMSIS is a fiendly ecosystem, it simplifies our work. Thank you.

Regards, 
Yuankang Li

closed 2022-03-18 by liyuankang

JonatanAntoni, 2022-03-28:
Hi @liyuankang,

You are welcome.
Please beware CoreValidation relies on Semihosting for `stdout` by default. This works fine when running the tests using Fast Models from command line but not within MDK. For MDK we have multiple options. One is configuring the project to retarget `stdout` to EventRecorder. 

Cheers,
Jonatan