ARM-software/CMSIS_5 #1453: Inconsistency between documentation and implementation in the CMSIS for the Cortex-M

PavolAstrab, 2022-03-31:
Inconsistency between documentation and implementation in the CMSIS for the Cortex-M
Can you check the inconsistency in static inline function `__USAT` parameters inside the file `cmsis_armcc.h`?
The first parameter is type `uint32_t`  (in documentation, See [CMSIS version 5.5](https://arm-software.github.io/CMSIS_5/Core/html/group__intrinsic__CPU__gr.html#ga76bbe4374a5912362866cdc1ded4064a)) :
```c
uint32_t __USAT  (  uint32_t  value, uint32_t sat )
```
but in the file `cmsis_armcc.h` the first parameter `val` is `int32_t` type:
```c
__attribute__((always_inline)) __STATIC_INLINE uint32_t __USAT(int32_t val, uint32_t sat)
{
... shortened
}
```
The same situation is in the CMSIS ver. 5.8.0

JonatanAntoni, 2022-03-31:
Hi @PavolAstrab,

thanks for pointing this out.

Cheers,
Jonatan

JonatanAntoni, 2022-03-31:
Hi @PavolAstrab,

The CMSIS definition in code seems to be in sync with the ACLE declaration (see https://developer.arm.com/documentation/101028/0013/?lang=en). Accordingly, the doc requires an update. I'll fix that.

Thanks again,
Jonatan

closed 2022-03-31 by JonatanAntoni

JonatanAntoni, 2022-03-31:
Updated documentation can be found at https://arm-software.github.io/CMSIS_5/develop/Core/html/group__intrinsic__CPU__gr.html#ga9ba87371aebd17dd6244ed3458b29b5d