ARM-software/CMSIS_5 #1457: Abysmal performance on arm_correlate_fast_*() functions

alireza-tabatabaee, 2022-04-06:
In my tests, I found out that the function _arm_correlate_fast_q15()_ on two 10,000 q15_t arrays on a Cortex-M7 processor takes 1,467,154,000 clocks to finish! 

I was able to achieve around 4 times less clocks (around 450,000,000 clocks) by implementing my own function based on the property that the fourier transform of the correlation of two signals is the conjugate of one multiplied by the other. F{corr(x,y)} = conjugate{F{x}} . F{y}

```
arm_cfft_f32(&arm_cfft_sR_f32_len32, a, 0, 1);
arm_cfft_f32(&arm_cfft_sR_f32_len32, b, 0, 1);
arm_cmplx_conj_f32(a, aconj, 32);
arm_mult_f32(aconj, b, c, 32);
arm_cfft_f32(&arm_cfft_sR_f32_len32, c, 1, 1);
arm_cmplx_mag_f32(c, out, 32);
return out;
```

Wouldn't it be more efficient for the correlate functions to use a fft based algorithm? After all, the philosophy of CMSIS DSP is to use the hardware to achieve maximum efficiency.

christophe0606, 2022-04-07:
@alireza-tabatabaee correlate / convolution functions are made for small vectors.
For big vectors it is always better to use FFT.

But for small vectors it would be too much. On some application you don't need the FFT tables to just implement a small correlate / convolution and also for very small vectors I don't think the FFT version would really be much faster.

That's why there are dedicated correlate / convolution functions.

alireza-tabatabaee, 2022-04-09:
Interesting, so could you please add this important information to the documentation?

christophe0606, 2022-04-11:
@alireza-tabatabaee Documentation will be updated in a future release.

christophe0606, 2022-06-30:
We are moving CMSIS-DSP to a new repository : https://github.com/ARM-software/CMSIS-DSP

I have recreated the github issues on the new repository (sometimes merging a few related ones). You can see the link just above this comment (... mentioned this issue ...).

So, I am closing the github issue here.

closed 2022-06-30 by christophe0606