ARM-software/CMSIS_5 #1464: PDSC validation checker incorrect

radhika-raghavendran, 2022-04-13:
I am generating a CMSIS pack for a new CM33 microcontroller from our group. During the PDSC schema validation a warning is issued for the release element of the PDSC. By checking with the schema file and the ARM website, the release element need not have a release date specified since release date is an optional field.
Attached is the error log as well as the ARM website information. 
[image]
[image]

jkrech, 2022-04-13:
a) the message type is INFO
b) you are correct that the release date is specified as optional. This attribute got added subsequently. For backward compatibility we are required to keep it optional in order for packs predating this update to the specification not to fail publishing.
c) we have decided to use PackChk in order to highlight to developers of packs that this is an important piece of information. 
Please note that PackChk is no longer maintained here. Please visit: 
Specification: https://github.com/Open-CMSIS-Pack/Open-CMSIS-Pack-Spec
PackChk tool: https://github.com/Open-CMSIS-Pack/devtools/

radhika-raghavendran, 2022-04-18:
Given that there are quite a few categories of messages that are optional but still flagged in the generation of the pack, what would be the right way to address them? What happens if we do not take care of these info messages while generating the pack?

jkrech, 2022-04-24:
Ideally you try to address all messages. If you find a message that you believe cannot be "addressed", I would be interested to know why and we may need to extend the documentation of the specification or the checking criteria accordingly.

radhika-raghavendran, 2022-04-24:
My question above was related to the initial report I had filed with you. The release date is an optional field but the new pack generator does generate a warning for it. We have currently chosen to ignore this warning because we will need to work back with our release tags to generate a date for each release. That is a lot of work at present!!
I would like to know a list of such warnings that which arise because of reasons that are not clearly explained in your documentation. If you could mention in your documentation of the PDSC regarding the optional fields which generate a warning, that would help us make an informed decision.