ARM-software/CMSIS_5 #1469: Compiling for cortex-m23 with armclang and armlink leads to an armlink error

grassead, 2022-04-27:
Hi all,

I am currently trying to build a simple program for the "FVP_MPS2_Cortex-M23" target.
This program only calls printf in its main function with a const string.

To be able to run of the target I also compile:
    - Device/ARM/ARMCM23/Source/system_ARMCM23.c
    - Device/ARM/ARMCM23/Source/ARM/startup_ARMCM23.S

Compilation option are: 
    * --target=arm-arm-none-eabi -mcpu=cortex-M23 -mfpu=none 
    * -DARMCM23 -DSTANDALONE 
    * -I"$(PACK)/CMSIS/Core/Include" -I"$(PACK)/Device/ARM/ARMCM23/Include" 
    * -O1 -g -MD -MP -c -fshort-enums -fshort-wchar  -Wno-dollar-in-identifier-extension -Wno-language-extension-token

I use armclang and armlink with version "ARM Compiler 6.6.4 Long Term Maintenance".
The .S is compiled with armclang and not armasm (as armasm complains about commentary and many other things).

When doing this, I get this error:

Error: L6238E: startup_ARMCM23.o(.text) contains invalid call from '~PRES8 (The user did not require code to preserve 8-byte aligment of 8-byte data objects)' function to 'REQ8 (Code was permitted to depend on the 8-byte aligment of 8-byte data items)' function SystemInit.

Proposed fix: Add ".eabi_attribute Tag_ABI_align_preserved, 1" at the beggining of "Device/ARM/ARMCM23/Source/ARM/startup_ARMCM23.S"

Thanks,

JonatanAntoni, 2022-04-27:
Hi @grassead,

This is an issue with Arm Compiler prior to 6.8.
https://developer.arm.com/documentation/101754/0618/armlink-Reference/armlink-Command-line-Options/--strict-preserve8-require8?lang=en

We never addressed this so far because the assembly startup code is considered deprecated in favor of the C startup code.
Let met check with the compiler experts about the recommended way to handle this.

Thanks,
Jonatan

grassead, 2022-04-27:
Hi @JonatanAntoni 

Thanks for the explanation.

I guess I should switch to the C startup code.
Let's try this for my case.

Feel free to close my MR and this bug if you will not fix the issue.

Thanks a lot.

JonatanAntoni, 2022-04-27:
We have basically two options if we want to use the old assembly startup with compilers prior to 6.8.
1. You can add `--diag_suppress=6238` to the linker flags to suppress the error.
2. We can add `.eabi_attribute Tag_ABI_align_preserved, 1` to all assembly files in gnu syntax.

Alternatively, you can try to use the C startup code. Or, if possible, update your Compiler to 6.16 LTS which has a lot of other issues fixed as well.

If you still think we should add the attribute, I don't have arguments against it. But we should add it to *all* assembly files in one go. Hence, I'd ask you to update your PR and add it everywhere.

Thanks,
Jonatan

JonatanAntoni, 2022-04-28:
We typically leave it to the requester to close tickets when one feels satisfied.

grassead, 2022-04-28:
Thanks a lot

closed 2022-04-28 by grassead