ARM-software/CMSIS_5 #1476: osThread callback

ThomasHohl, 2022-05-04:
We are currently moving code from C to C++ and one part of it is to create a wrapper class that encapsulates the Keil RTOS2 API (RTX V5.5.3) C functions. Method start() calls the RTX osThreadNew() function. As the first parameter of osThreadNew() the thread function is passed. In our case member function keilFunction() shall be passed.

```
iOS::STATUS iOS_Thread::start( void )
{
  auto returnValue = iOS::STATUS::ERROR;

  iOS_THREAD_KEIL_FUNCTION<void( void * )>::func = std::bind( &iOS_Thread::keilFunction, this, std::placeholders::_1 );
  keilFunction_t func = static_cast<keilFunction_t>( iOS_THREAD_KEIL_FUNCTION<void( void * )>::keilFunction );

  m_osThreadId = osThreadNew( func, nullptr, &m_osThreadAttributes );

  IBASE_ASSERT( nullptr != m_osThreadId );

  if ( nullptr == m_osThreadId )
  {
    goto error;
  }
  
  returnValue = iOS::STATUS::OK;

error:

  return ( returnValue );

} /* start() */

void iOS_Thread::keilFunction( void *argument )
{
  IBASE_UNUSED( argument );

  run( m_argument );

} /* keilFunction() */
```

In an example two threads threadA and threadB are created and started. When debugging, it seems that only thread B is executed. Well actually both threads A and B are executed, but both call the thread function keilFunction() of the same object (threadB).

If the code of start() and keilFunction() is modified by adding a semaphore used to block the creation of a new iOS_Thread object until the keilFunction has been entered, everything works as expected.
```
iOS::STATUS iOS_Thread::start( void )
{
  auto returnValue = iOS::STATUS::ERROR;

  iOS_THREAD_KEIL_FUNCTION<void( void * )>::func = std::bind( &iOS_Thread::keilFunction, this, std::placeholders::_1 );
  keilFunction_t func = static_cast<keilFunction_t>( iOS_THREAD_KEIL_FUNCTION<void( void * )>::keilFunction );

  m_osSemaphoreId = osSemaphoreNew( 1, 0, nullptr );
  
  m_osThreadId = osThreadNew( func, nullptr, &m_osThreadAttributes );

  IBASE_ASSERT( nullptr != m_osThreadId );

  if ( nullptr == m_osThreadId )
  {
    goto error;
  }

  osSemaphoreAcquire( m_osSemaphoreId, osWaitForever );
  osSemaphoreDelete( m_osSemaphoreId );
  
  returnValue = iOS::STATUS::OK;

error:

  return ( returnValue );

} /* start() */

void iOS_Thread::keilFunction( void *argument )
{
  IBASE_UNUSED( argument );

  osSemaphoreRelease( m_osSemaphoreId );
  
  run( m_argument );

} /* keilFunction() */
```

The workaround with the semaphore seems to work, however I would like to understand what happens and if this is a suitable solution.

JonatanAntoni, 2022-05-04:
Hi Thomas,

Thanks for reaching out. To be honest, we have C++ support on our backlog but actually didn't think much about it, yet.

I guess the issue in your code is that you pass a stack object as the function to `osThreadNew`.
Using the semaphore you expand the stack objects live time until after the thread is running.

I am used to another technique:
```c++
class Thread {
  /* trampoline */
  static void keilFunction(void* arg) {
    static_cast<Thread*>(arg)->execute();
  }

  virtual void run() = 0;

  void start() {
    m_osThreadId = osThreadNew(Thread::keilFunction, this, &m_osThreadAttributes);
  }
}
```

I agree, using `std::bind` looks like more modern C++. What happens if you store `func` as a member `m_func` instead of keeping it on the callstack of `start()`?

Cheers,
Jonatan

ThomasHohl, 2022-05-04:
Hi Jonatan

Thank you very much for your quick reply.

Storing `func` as a member does not solve the problem, however the example you provided works like a charm and is far more understandable than the `std::bind` thing.

Thank you,
Thomas

closed 2022-05-04 by ThomasHohl