ARM-software/CMSIS_5 #1478: Link error with CMSIS 5.9.0 (CMSIS-DSP 1.10.0)

escherstair, 2022-05-05:
After I upgraded from CMSIS 5.8.0 to 5.9.0 I get the following link error
`Undefined symbol arm_sqrt_q15 (referred from complexmathfunctions.o).`
To optimize the space I use proper defines to include the needed tables only, as described [here](https://github.com/ARM-software/CMSIS_5/tree/develop/CMSIS/DSP#compilation-symbols-for-tables).

Here is the list of my defines
```
ARM_MATH_LOOPUNROLL
DISABLEFLOAT16
ARM_DSP_CONFIG_TABLES
ARM_FAST_ALLOW_TABLES
ARM_FFT_ALLOW_TABLES
ARM_TABLE_SIN_F32
ARM_TABLE_TWIDDLECOEF_F32_2048
ARM_TABLE_BITREVIDX_FLT_2048
ARM_TABLE_TWIDDLECOEF_RFFT_F32_4096
```

The problem is that here
https://github.com/ARM-software/CMSIS_5/blob/3a0dc602aeead14363bdfc9c0157977007c9e443/CMSIS/DSP/Source/FastMathFunctions/FastMathFunctions.c#L59-L61
`arm_sqrt_q15.c` is not included.
But, as an example, in several lines of [this file](https://github.com/ARM-software/CMSIS_5/blob/develop/CMSIS/DSP/Source/ComplexMathFunctions/arm_cmplx_mag_fast_q15.c), `arm_sqrt_q15()` is called without any guards.

christophe0606, 2022-05-06:
@escherstair You need to add a define for `ARM_TABLE_SQRT_Q15`

`ARM_FAST_ALLOW_TABLES` only means that some code needing interpolation table is used. But then you need to specify which interpolation table is used either all with `ARM_ALL_FAST_TABLES` or only the sqrt with `ARM_TABLE_SQRT_Q15`

Those settings are low level and not very easy to use. At some point I intend to add another layer to make it easier for developers.

You can also use the Python script DSP/cmsisdspconfig.py

```
pip install streamlit
streamlit run cmsisdspconfig.py
```

It will generate the define you need.

escherstair, 2022-05-06:
Hello @christophe0606,
I don't understand what you mean.
I generated the proper defines with the Python script DSP/cmsisdspconfig.py using CMSIS 5.8.0 (DSP 1.9.0) and it worked.
I don't need and I don't use q15 math in my application.
Here is how I configured the script
[image]
[image]
[image]
[image]

After I upgraded CMSIS to 5.9.0 (and so DSP to 1.10.0), I saw that the script has been updated (for "Reworked sqrt q15 and q31") and so I used the new script to generate the updated defines.
But the same defines have been generated.
The problem is that with the updated CMSIS/DSP `arm_sqrt_q15` is excluded from the buils (this is ok with my defines), but some code (not excluded) refer to this function.
I think some define guards are missing in the reworked code.

christophe0606, 2022-05-06:
@escherstair I understand now. Thanks for the clarification.

So it indeed looks like some new functions have been added which are dependent on sqrt_q15 and are not taken into account by the defines.

I think you can still just add arm_sqrt_q15 even if you don't use it. The linker should be able to remove this unused code and related tables (it is not always possible when some initialization functions are required but in case of sqrt there is none).

I'll improve this with new #define in a future commit.

escherstair, 2022-05-06:
@christophe0606 I think the reason is that the ikssue comes from [this commit](https://github.com/ARM-software/CMSIS_5/commit/d7b6a57beb97c00f67e001c8a71fa9d672251eb9#diff-6201c16a547b60f1a7342c5943dddecb80eb30d1905c8ded9ccd56203f15484f).
Before it, `arm_sqrt_q15.c` was always included (for every combination of the defines).
Now it's included only
`#if !defined(ARM_DSP_CONFIG_TABLES) || defined(ARM_ALL_FAST_TABLES) || defined(ARM_TABLE_SQRT_Q15)`

christophe0606, 2022-05-06:
@escherstair It is because the square root implementation has been improved and is using a lookup table.
So, it was normal to introduce those new defines but then we should have tracked all the functions which are using those square roots because they now also have a dependency on those new lookup tables.

So, it is something which must be corrected. Thanks for reporting it.

christophe0606, 2022-05-16:
@escherstair I switch the tag to enhancement because I need to rethink all of this configuration code because it is getting too complex and difficult to maintain so there will probably be other problems like the one you reported.

So instead of fixing just this issue I want to be sure there won't be other ones.

christophe0606, 2022-06-30:
We are moving CMSIS-DSP to a new repository : https://github.com/ARM-software/CMSIS-DSP

I have recreated the github issues on the new repository (sometimes merging a few related ones). You can see the link just above this comment (... mentioned this issue ...).

So, I am closing the github issue here.

closed 2022-06-30 by christophe0606