ARM-software/CMSIS_5 #1484: cpp code include cmsis_compiler.h will issue -fpermissive error on arm-none-eabi-g++ 2021.10

versaloon, 2022-05-12:
source code is simple:
```cpp
#include <cmsis_compiler.h>

void * alloca(int size)
{
    return (void *)0;
}
```

arm-none-eabi-g++ -v output:
```
Using built-in specs.
COLLECT_GCC=C:\Program Files (x86)\GNU Arm Embedded Toolchain\10 2021.10\bin\arm-none-eabi-g++.exe
COLLECT_LTO_WRAPPER=c:/program\ files\ (x86)/gnu\ arm\ embedded\ toolchain/10\ 2021.10/bin/../lib/gcc/arm-none-eabi/10.3.1/lto-wrapper.exe
Target: arm-none-eabi
Configured with: /mnt/workspace/workspace/GCC-10-pipeline/jenkins-GCC-10-pipeline-338_20211018_1634516203/src/gcc/configure --build=x86_64-linux-gnu --host=i686-w64-mingw32 --target=arm-none-eabi --prefix=/mnt/workspace/workspace/GCC-10-pipeline/jenkins-GCC-10-pipeline-338_20211018_1634516203/install-mingw --libexecdir=/mnt/workspace/workspace/GCC-10-pipeline/jenkins-GCC-10-pipeline-338_20211018_1634516203/install-mingw/lib --infodir=/mnt/workspace/workspace/GCC-10-pipeline/jenkins-GCC-10-pipeline-338_20211018_1634516203/install-mingw/share/doc/gcc-arm-none-eabi/info --mandir=/mnt/workspace/workspace/GCC-10-pipeline/jenkins-GCC-10-pipeline-338_20211018_1634516203/install-mingw/share/doc/gcc-arm-none-eabi/man --htmldir=/mnt/workspace/workspace/GCC-10-pipeline/jenkins-GCC-10-pipeline-338_20211018_1634516203/install-mingw/share/doc/gcc-arm-none-eabi/html --pdfdir=/mnt/workspace/workspace/GCC-10-pipeline/jenkins-GCC-10-pipeline-338_20211018_1634516203/install-mingw/share/doc/gcc-arm-none-eabi/pdf --enable-languages=c,c++ --enable-mingw-wildcard --disable-decimal-float --disable-libffi --disable-libgomp --disable-libmudflap --disable-libquadmath --disable-libssp --disable-libstdcxx-pch --disable-nls --disable-shared --disable-threads --disable-tls --with-gnu-as --with-gnu-ld --with-headers=yes --with-newlib --with-python-dir=share/gcc-arm-none-eabi --with-sysroot=/mnt/workspace/workspace/GCC-10-pipeline/jenkins-GCC-10-pipeline-338_20211018_1634516203/install-mingw/arm-none-eabi --with-libiconv-prefix=/mnt/workspace/workspace/GCC-10-pipeline/jenkins-GCC-10-pipeline-338_20211018_1634516203/build-mingw/host-libs/usr --with-gmp=/mnt/workspace/workspace/GCC-10-pipeline/jenkins-GCC-10-pipeline-338_20211018_1634516203/build-mingw/host-libs/usr --with-mpfr=/mnt/workspace/workspace/GCC-10-pipeline/jenkins-GCC-10-pipeline-338_20211018_1634516203/build-mingw/host-libs/usr --with-mpc=/mnt/workspace/workspace/GCC-10-pipeline/jenkins-GCC-10-pipeline-338_20211018_1634516203/build-mingw/host-libs/usr --with-isl=/mnt/workspace/workspace/GCC-10-pipeline/jenkins-GCC-10-pipeline-338_20211018_1634516203/build-mingw/host-libs/usr --with-libelf=/mnt/workspace/workspace/GCC-10-pipeline/jenkins-GCC-10-pipeline-338_20211018_1634516203/build-mingw/host-libs/usr --with-host-libstdcxx='-static-libgcc -Wl,-Bstatic,-lstdc++,-Bdynamic -lm' --with-pkgversion='GNU Arm Embedded Toolchain 10.3-2021.10' --with-multilib-list=rmprofile,aprofile
Thread model: single
Supported LTO compression algorithms: zlib
gcc version 10.3.1 20210824 (release) (GNU Arm Embedded Toolchain 10.3-2021.10)
```

compiler error message:
```
In file included from C:/Users/simon/Desktop/vsf/vsf.linux/vsf.linux/vsf/source/utilities/compiler/arm/3rd-party/CMSIS/CMSIS/Core/Include/cmsis_compiler.h:54,
                 from C:/Users/simon/Desktop/vsf/vsf.linux/vsf.linux/application/test.cpp:1:
C:/Users/simon/Desktop/vsf/vsf.linux/vsf.linux/vsf/source/utilities/compiler/arm/3rd-party/CMSIS/CMSIS/Core/Include/cmsis_gcc.h: In function 'void __cmsis_start()':
C:/Users/simon/Desktop/vsf/vsf.linux/vsf.linux/vsf/source/utilities/compiler/arm/3rd-party/CMSIS/CMSIS/Core/Include/cmsis_gcc.h:146:31: error: 'const __cmsis_start()::__copy_table_t __copy_table_start__', declared using local type 'const __cmsis_start()::__copy_table_t', is used but never defined [-fpermissive]
  146 |   extern const __copy_table_t __copy_table_start__;
      |                               ^~~~~~~~~~~~~~~~~~~~
C:/Users/simon/Desktop/vsf/vsf.linux/vsf.linux/vsf/source/utilities/compiler/arm/3rd-party/CMSIS/CMSIS/Core/Include/cmsis_gcc.h:147:31: error: 'const __cmsis_start()::__copy_table_t __copy_table_end__', declared using local type 'const __cmsis_start()::__copy_table_t', is used but never defined [-fpermissive]
  147 |   extern const __copy_table_t __copy_table_end__;
      |                               ^~~~~~~~~~~~~~~~~~
C:/Users/simon/Desktop/vsf/vsf.linux/vsf.linux/vsf/source/utilities/compiler/arm/3rd-party/CMSIS/CMSIS/Core/Include/cmsis_gcc.h:148:31: error: 'const __cmsis_start()::__zero_table_t __zero_table_start__', declared using local type 'const __cmsis_start()::__zero_table_t', is used but never defined [-fpermissive]
  148 |   extern const __zero_table_t __zero_table_start__;
      |                               ^~~~~~~~~~~~~~~~~~~~
C:/Users/simon/Desktop/vsf/vsf.linux/vsf.linux/vsf/source/utilities/compiler/arm/3rd-party/CMSIS/CMSIS/Core/Include/cmsis_gcc.h:149:31: error: 'const __cmsis_start()::__zero_table_t __zero_table_end__', declared using local type 'const __cmsis_start()::__zero_table_t', is used but never defined [-fpermissive]
  149 |   extern const __zero_table_t __zero_table_end__;
```

versaloon, 2022-05-12:
If cpp source codes are removed, comile will be OK.

JonatanAntoni, 2022-05-12:
Hi @versaloon,

This seems to be a duplicate of #617.
Unfortunately, we don't have a satisfying solution, yet.

Cheers,
Jonatan