ARM-software/CMSIS_5 #1489: Issue with Cmake on Apple Silicone

Silfurion, 2022-05-18:
Hello Everyone , 
I'm working on Apple Silicone (Apple Mini with Apple M1 ) with the version 12.3.1 of macOS and 13.4 of Xcode. 
I want to create a macOS Apple Silicone Application which will use CMSIS library (only the DSP part ). My goal is to compile CMSIS for ARM64 on macOS with Xcode but I have problem when I want to use cmake to create the Xcode project. 

I tried to only use cmake the dsp part of the library following instructions in the CMSIS/DSP/Readme file. 
So I created a folder named OnlyDsp , on this folder I create the following CMakeLists.txt (exactly the same as in the Readme) 
[CMakeLists.txt](https://github.com/ARM-software/CMSIS_5/files/8715111/CMakeLists.txt)
Next , I created a build folder where I put the following cmake command : 

cmake -DCMAKE_TOOLCHAIN_FILE="../../armac6.cmake"  -DROOT="../../../../" -DPLATFORM="FVP" -G "Xcode" ..

and I have the following error : 
[CMakeError.log](https://github.com/ARM-software/CMSIS_5/files/8715176/CMakeError.log)

You can find here the Outputlog to : 
[CMakeOutput.log](https://github.com/ARM-software/CMSIS_5/files/8715179/CMakeOutput.log)

Does someone knows how to avoid this issue?

christophe0606, 2022-05-19:
@Silfurion macOS is not part of our supported platforms : which means we do not test very often. But some people managed to build on it.

I think you should remove the option -DCMAKE_TOOLCHAIN_FILE which is only useful for cross-compilation.

And you may also add : -DNEON=ON
If you have additional problems with float16, you can disable this part with:
-DDISABLEFLOAT16=ON

christophe0606, 2022-06-30:
The README on the new CMSIS-DSP repository has a section about building with cmake and aarch64:

https://github.com/ARM-software/CMSIS-DSP

Note that you can use the CMakeLists.txt from the DSP/Source folder. But the other cmake files from Testing should not be used.

If you still have problems, open a new issue on the new repository.

closed 2022-06-30 by christophe0606