ARM-software/CMSIS_5 #1504: CMSIS-DSP：result of function arm_mat_cmplx_mult_q31 maybe incorrect

JoleenSun, 2022-06-14:
`        while (colCnt > 0U)
        {
          /* c(m,n) = a(1,1) * b(1,1) + a(1,2) * b(2,1) + .... + a(m,p) * b(p,n) */
          a1 = *(pIn1     );
          c1 = *(pIn2     );
          b1 = *(pIn1 + 1U);
          d1 = *(pIn2 + 1U);

          /* Multiply and Accumlates */
          sumReal += (q63_t) a1 * c1;
          sumImag += (q63_t) b1 * c1;

          /* update pointers */
          pIn1 += 2U;
          pIn2 += 2 * numColsB;

          /* Multiply and Accumlates */
          sumReal -= (q63_t) b1 * d1;
          sumImag += (q63_t) a1 * d1;

          /* Decrement loop counter */
          colCnt--;
        }

        /* Store result in destination buffer */
        *px++ = (q31_t) clip_q63_to_q31(sumReal >> 31);
        *px++ = (q31_t) clip_q63_to_q31(sumImag >> 31);`

when the result of the multiply-accumulate for the sumImag(q63_t) overflows(value of the input arraies are bit enough or size of the array is big ), before the clip_q63_to_q31  the sumImag has already been cut short for q63_t type(clip_q63_to_q31 seems too late to do the saturation), which may lead to  sign reversal(turn into  negative) or non-saturated value(still positive), but in fact, it should be a saturated value.

christophe0606, 2022-06-14:
@JoleenSun Thank for asking. It is detailed in the documentation:

> The function is implemented using an internal 64-bit accumulator.
>                    The accumulator has a 2.62 format and maintains full precision of the intermediate
>                    multiplication results but provides only a single guard bit. There is no saturation
>                    on intermediate additions. Thus, if the accumulator overflows it wraps around and
>                    distorts the result. The input signals should be scaled down to avoid intermediate
>                    overflows. The input is thus scaled down by log2(numColsA) bits
>                    to avoid overflows, as a total of numColsA additions are performed internally.
>                    The 2.62 accumulator is right shifted by 31 bits and saturated to 1.31 format to yield the final result.

So in fact the latest `clip_q63_to_q31` is only useful in the very limited use case where we have a value < 2 in absolute value for the sum.

JoleenSun, 2022-06-17:
Thanks for your answer. Then above limitations should  be taken into account when create test arrays. Get it, thanks a lot!

closed 2022-06-17 by JoleenSun