ARM-software/CMSIS_5 #1507: ETA for aarch64 DSP support

bobkocisko, 2022-06-17:
Hello!  I've seen multiple issues (#983, #1396) regarding support for Cortex-A53 but so far I haven't seen any commitment regarding a timeline for updating the DSP or core libraries to support it.  In particular we are hoping to use the DSP library for programming Cortex-A76 and Cortex-A55 and are wondering when that might be possible?

christophe0606, 2022-06-20:
Hello @bobkocisko,

CMSIS-DSP started as a Cortex-M library and we are **slowly** adding more Cortex-A support. Currently there is nearly no aarch64 support.

I can't communicate any schedule. But we do have some plan for more aarch64 support.

christophe0606, 2022-06-20:
@bobkocisko If you have specific need in mind (some specific functions) then it may help us to prioritize and know which functions to do first.

bobkocisko, 2022-06-20:
Thanks @christophe0606!  Sure, these are the specific functions we're currently interested in, in priority order:

arm_add_f32
arm_mult_f32
arm_biquad_cascade_df2T_*_f32
arm_power_f32
arm_cos_f32
arm_sin_f32
arm_vlog_f32
arm_sqrt_f32
arm_rms_f32

bobkocisko, 2022-06-20:
A related question:  Is it possible to compile the DSP library for the Cortex-A32 and then run it on the A76 and A55 in aarch32 mode?  If so, what optimizations would we lose?

christophe0606, 2022-06-21:
@bobkocisko  I am not knowledgeable enough in aarch64. I know that, for our dot product, the neon implementation is different for aarch64 and we are using the `vpadds_f32` instruction not available in aarch32. It gives some performance improvement.

There are more instructions for aarch64 and some of them may be useful for optimizing some kernels. But it is highly kernel dependent and more likely for big kernels that the ones you listed above.

You can build CMSIS-DSP in aarch64 today. We use instructions available for both aarch32 and aarch64 (except for the dot product where we have an aarch64 customization). So some additional optimizations may be possible with specific aarch64 instructions. 

Since you have more registers in aarch64, you'll benefit from compiling for aarch64.

The functions from your list are already using Neon instructions (when it is making sense). I think rms could be slightly optimized by also using the `vpadds_f32` like our dot product.

bobkocisko, 2022-06-21:
Good to know!  What would we use for -DARM_CPU="???" in the cmake build command when building just the DSP?  I followed the instructions in DSP/README.md but am still not sure about that.  Would it be "aarch64"?  Also, we have an SDK from Buildroot for our target machine, should we be referencing that toolchain or should we be trying to use the toolchain provided in the DSP/Toolchain folder?

christophe0606, 2022-06-22:
@bobkocisko It is not going to work with the current cmake files unfortunately. Those files were initially developed for our internal test framework and only support a subset of the Cortex-A cores.

So you'll need to create your cmake or Makefile. It is simpler than it looks. For instance, in DSP/Source/BasicMathFunctions you'll see a file BasicMathFunctions.c.

Just use it. It is including all other basic math C sources. So your cmake or Makefile can only use those files.

In a future release, it is planned to clean the cmake so that the dependencies to the test framework are removed which would make it easier to build CMSIS-DSP only.

Then, for the toolchain : you can try with the gcc.cmake file included.

Now for your test with the sdk, `compilerSpecificCompileOptions` is defined in DSP/Toolchain/Tools. And it is included by `configLib`.

It looks like it was not included. Perhaps it is due to the fact that the current framework is supporting only a subset of the Cortex-A core and your Cortex-A is not recognized.

I think it is better you avoid using current cmake files. You'll have better control on the compilation options which must be used for your Cortex-A (fpu etc ...) if you write your own cmake file or Makefile.

Some people are building CMSIS-DSP on Windows on Arm and on the Mac M1. 
So it can definitely be built on aarch64 but it still requires a bit more work to do it.

bobkocisko, 2022-06-24:
Thanks for the suggestion to make my own CMakeFiles.txt, that did the trick.  I was able to figure out what the issues were and then tweaked an existing .cmake file and now can build the whole library within the existing framework.  Yay!  

I did need to `target_include_directories(${PROJECTNAME} PUBLIC "${ROOT}/CMSIS/Core/Include")` for the compilation to run; I first tried `target_include_directories(${PROJECTNAME} PUBLIC "${ROOT}/CMSIS/Core_A/Include")`, but that caused issues with duplicate symbols.  Anyway, do you see any problem with this dependency on the Core includes even though we're only doing DSP?

One issue I'm coming across is that when I build with -DNEON=ON I get a ton of warnings like this:

```
[ 14%] Building C object ComplexMathFunctions/CMakeFiles/CMSISDSPComplexMath.dir/arm_cmplx_mag_f32.c.o
In file included from /home/sandal/CMSIS_5/CMSIS/DSP/Source/build/../../../../CMSIS/DSP/Include/arm_vec_math.h:304,
                 from /home/sandal/CMSIS_5/CMSIS/DSP/Source/ComplexMathFunctions/arm_cmplx_mag_f32.c:73:
/home/sandal/CMSIS_5/CMSIS/DSP/Source/build/../../../../CMSIS/DSP/ComputeLibrary/Include/NEMath.h:394:79: warning: ‘vlogq_f32’ is static but used in inline function ‘vlogq_f16’ which is not static
  394 |     const float16x8_t res = vcvt_high_f16_f32(vcvt_f16_f32(vlogq_f32(x_low)), vlogq_f32(x_high));
      |                                                                               ^~~~~~~~~
/home/sandal/CMSIS_5/CMSIS/DSP/Source/build/../../../../CMSIS/DSP/ComputeLibrary/Include/NEMath.h:394:60: warning: ‘vlogq_f32’ is static but used in inline function ‘vlogq_f16’ which is not static
  394 |     const float16x8_t res = vcvt_high_f16_f32(vcvt_f16_f32(vlogq_f32(x_low)), vlogq_f32(x_high));
      |
```

Is this anything to be concerned about?

Also, I'm not sure which compilation options should be used for the cortex-a76.cortex-a55 chips, are there any reference documents for that or do I need to piece it together by looking at the spec sheets and connect the dots there?

christophe0606, 2022-06-27:
@bobkocisko There is a dependency to Core for two reasons:

- Compiler macros (static inline, alignment ...) which enable more portability of the library 
- DSP intrinsics (CLZ, SSAT ...) used in scalar code. In theory they are no more needed since compilers are now implementing the ACLE spec (Arm C Language extensions) and those macros are no more needed. But CMSIS-DSP, for historical reason, is still using those old definitions.

I am concerned that you was not able to use `Core_A/Include` and had to use `Core/Include`. `Core/Include` is for Cortex-M and although there is probably an overlap between both, you should really use Core_A.

What are the duplicate symbols you get ?

Regarding the warning message with `vlogq_f32` : indeed in the same header file the function is declared with static inline and defined with only inline.
It is not correct (but won't cause you any problem). What are the compilation options you are using ? I have enabled most warnings and never got this one.

Finally, regarding the options to build for Cortex-A : I don't know. You need to specify the fpu. In gcc, there is an option to list the fpu available when you have chosen the core.

bobkocisko, 2022-06-27:
I created a fork so you can see exactly what I'm doing on my end.

Here is the commit and in particular the specific line in question where I'm referencing Core_A/Include:
https://github.com/bobkocisko/CMSIS_5/commit/140bf5195baa47df19438236ac0c351a3c526f61#diff-477c527dfcaf58a0cf35406c7620a9ae88820fdf17d13f0b64e28c30173f5b03R37

Now for the full picture of what I'm doing.

First, I'm sourcing the environment provided by the Buildroot-based sdk
```
$ source ~/aarch64-buildroot-linux-gnu_sdk-buildroot/environment-setup
 _           _ _     _                 _
| |__  _   _(_) | __| |_ __ ___   ___ | |_
| '_ \| | | | | |/ _` | '__/ _ \ / _ \| __|
| |_) | |_| | | | (_| | | | (_) | (_) | |_
|_.__/ \__,_|_|_|\__,_|_|  \___/ \___/ \__|

       Making embedded Linux easy!

Some tips:
* PATH now contains the SDK utilities
* Standard autotools variables (CC, LD, CFLAGS) are exported
* Kernel compilation variables (ARCH, CROSS_COMPILE, KERNELDIR) are exported
* To configure do "./configure $CONFIGURE_FLAGS" or use
  the "configure" alias
* To build CMake-based projects, use the "cmake" alias 
```

which sets up this alias
```
$ alias cmake
alias cmake='cmake -DCMAKE_TOOLCHAIN_FILE=/home/sandal/aarch64-buildroot-linux-gnu_sdk-buildroot/share/buildroot/toolchainfile.cmake -DCMAKE_INSTALL_PREFIX=/usr'
```
which references `toolchainfile.cmake`
```cmake
#
# Automatically generated file; DO NOT EDIT.
# CMake toolchain file for Buildroot
#

# In order to allow the toolchain to be relocated, we calculate the
# HOST_DIR based on this file's location: $(HOST_DIR)/share/buildroot
# and store it in RELOCATED_HOST_DIR.
# All the other variables that need to refer to HOST_DIR will use the
# RELOCATED_HOST_DIR variable.
string(REPLACE "/share/buildroot" "" RELOCATED_HOST_DIR ${CMAKE_CURRENT_LIST_DIR})

# Point cmake to the location where we have our custom modules,
# so that it can find our custom platform description.
list(APPEND CMAKE_MODULE_PATH ${CMAKE_CURRENT_LIST_DIR})

set(CMAKE_SYSTEM_NAME Buildroot)
set(CMAKE_SYSTEM_VERSION 1)
set(CMAKE_SYSTEM_PROCESSOR aarch64)

# Set the {C,CXX}FLAGS appended by CMake depending on the build type
# defined by Buildroot. CMake defaults these variables with -g and/or
# -O options, and they are appended at the end of the argument list,
# so the Buildroot options are overridden. Therefore these variables
# have to be cleared, so that the options passed in CMAKE_C_FLAGS do
# apply.
#
# Note:
#   if the project forces some of these flag variables, Buildroot is
#   screwed up and there is nothing Buildroot can do about that :(
set(CMAKE_C_FLAGS_DEBUG "" CACHE STRING "Debug CFLAGS")
set(CMAKE_C_FLAGS_RELEASE " -DNDEBUG" CACHE STRING "Release CFLAGS")

# Build type from the Buildroot configuration
set(CMAKE_BUILD_TYPE Release CACHE STRING "Buildroot build configuration")

[... 379 lines cut ...]
In file included from /home/sandal/CMSIS_5_BK/CMSIS/DSP/Source/build/../../../../CMSIS/Core_A/Include/cmsis_compiler.h:48,
                 from /home/sandal/CMSIS_5_BK/CMSIS/DSP/Source/build/../../../../CMSIS/DSP/Include/arm_math_types.h:82,
                 from /home/sandal/CMSIS_5_BK/CMSIS/DSP/Source/build/../../../../CMSIS/DSP/Include/dsp/basic_math_functions.h:30,
                 from /home/sandal/CMSIS_5_BK/CMSIS/DSP/Source/BasicMathFunctions/arm_abs_f64.c:29:
/home/sandal/CMSIS_5_BK/CMSIS/DSP/Source/build/../../../../CMSIS/Core_A/Include/cmsis_gcc.h:195:31: note: previous definition of ‘__SXTB16’ was here
  195 | __STATIC_FORCEINLINE uint32_t __SXTB16(uint32_t op1)
      |                               ^~~~~~~~
In file included from /home/sandal/CMSIS_5_BK/CMSIS/DSP/Source/build/../../../../CMSIS/DSP/Include/dsp/basic_math_functions.h:33,
                 from /home/sandal/CMSIS_5_BK/CMSIS/DSP/Source/BasicMathFunctions/arm_abs_f64.c:29:
/home/sandal/CMSIS_5_BK/CMSIS/DSP/Source/build/../../../../CMSIS/DSP/Include/dsp/none.h:561:32: error: redefinition of ‘__SMMLA’
  561 |   __STATIC_FORCEINLINE int32_t __SMMLA(
      |                                ^~~~~~~
In file included from /home/sandal/CMSIS_5_BK/CMSIS/DSP/Source/build/../../../../CMSIS/Core_A/Include/cmsis_compiler.h:48,
                 from /home/sandal/CMSIS_5_BK/CMSIS/DSP/Source/build/../../../../CMSIS/DSP/Include/arm_math_types.h:82,
                 from /home/sandal/CMSIS_5_BK/CMSIS/DSP/Source/build/../../../../CMSIS/DSP/Include/dsp/basic_math_functions.h:30,
                 from /home/sandal/CMSIS_5_BK/CMSIS/DSP/Source/BasicMathFunctions/arm_abs_f64.c:29:
/home/sandal/CMSIS_5_BK/CMSIS/DSP/Source/build/../../../../CMSIS/Core_A/Include/cmsis_gcc.h:261:30: note: previous definition of ‘__SMMLA’ was here
  261 | __STATIC_FORCEINLINE int32_t __SMMLA (int32_t op1, int32_t op2, int32_t op3)
      |                              ^~~~~~~
make[2]: *** [BasicMathFunctions/CMakeFiles/CMSISDSPBasicMath.dir/build.make:76: BasicMathFunctions/CMakeFiles/CMSISDSPBasicMath.dir/arm_abs_f64.c.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:350: BasicMathFunctions/CMakeFiles/CMSISDSPBasicMath.dir/all] Error 2
make: *** [Makefile:91: all] Error 2
```

If on the other hand I switch from `Core_A/Include` to `Core/Include` then everything compiles fine but I get all the warnings I mentioned in the last message.

bobkocisko, 2022-06-27:
One last perhaps useful detail, here is the gcc version
```
$ $(echo $CC) --version
aarch64-buildroot-linux-gnu-gcc.br_real (Buildroot -gc7690e53f5-dirty) 10.3.0
Copyright (C) 2020 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```

christophe0606, 2022-06-28:
@bobkocisko It looks like `__ARM_FEATURE_DSP` is not defined when compiling for your Cortex-A target.

But there is no check for `__ARM_FEATURE_DSP` in Core_A so intrinsics for `__SXTB16` for instance is always defined here.

CMSIS-DSP is also redefining `__SXTB16` in `dsp/none.h` since the flag `ARM_MATH_DSP` is not defined because `__ARM_FEATURE_DSP` is not defined.

I believe that those intrinsics are always available on Cortex-A and no need to check for `__ARM_FEATURE_DSP`.

So, you can force CMSIS-DSP by defining `-DARM_MATH_DSP` when building.
It will tell CMSIS-DSP that DSP extensions are available and CMSIS-DSP will not try to include a C equivalent implementation. It will remove the conflict with Core_A

bobkocisko, 2022-06-28:
Progress!  You can see my updated configCore.cmake [here](https://github.com/bobkocisko/CMSIS_5/commit/e66b6f3f05a62708b6d5d7dd393d37018b4eefa5#diff-477c527dfcaf58a0cf35406c7620a9ae88820fdf17d13f0b64e28c30173f5b03R38)

Here is the next problem

```
$ cmake -DROOT="$(pwd)/../../../../" -DARM_CPU="cortex-a76.cortex-a55" -DGCC=ON -DNEON=ON -G "Unix Makefiles" ..
-- The C compiler identification is GNU 10.3.0
-- The CXX compiler identification is GNU 10.3.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /home/sandal/aarch64-buildroot-linux-gnu_sdk-buildroot/bin/aarch64-buildroot-linux-gnu-gcc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /home/sandal/aarch64-buildroot-linux-gnu_sdk-buildroot/bin/aarch64-buildroot-linux-gnu-g++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- The ASM compiler identification is GNU
-- Found assembler: /home/sandal/aarch64-buildroot-linux-gnu_sdk-buildroot/bin/aarch64-buildroot-linux-gnu-gcc
-- Configuring done
-- Generating done
-- Build files have been written to: /home/sandal/CMSIS_5_BK/CMSIS/DSP/Source/build
$ make
[  0%] Building C object BasicMathFunctions/CMakeFiles/CMSISDSPBasicMath.dir/arm_abs_f64.c.o
[  0%] Building C object BasicMathFunctions/CMakeFiles/CMSISDSPBasicMath.dir/arm_add_f64.c.o
[  0%] Building C object BasicMathFunctions/CMakeFiles/CMSISDSPBasicMath.dir/arm_dot_prod_f64.c.o
[  0%] Building C object BasicMathFunctions/CMakeFiles/CMSISDSPBasicMath.dir/arm_mult_f64.c.o
[  0%] Building C object BasicMathFunctions/CMakeFiles/CMSISDSPBasicMath.dir/arm_negate_f64.c.o
[  1%] Building C object BasicMathFunctions/CMakeFiles/CMSISDSPBasicMath.dir/arm_offset_f64.c.o
[  1%] Building C object BasicMathFunctions/CMakeFiles/CMSISDSPBasicMath.dir/arm_scale_f64.c.o
[  1%] Building C object BasicMathFunctions/CMakeFiles/CMSISDSPBasicMath.dir/arm_sub_f64.c.o
[  1%] Building C object BasicMathFunctions/CMakeFiles/CMSISDSPBasicMath.dir/arm_abs_f32.c.o
[  1%] Building C object BasicMathFunctions/CMakeFiles/CMSISDSPBasicMath.dir/arm_add_f32.c.o
[  1%] Building C object BasicMathFunctions/CMakeFiles/CMSISDSPBasicMath.dir/arm_clip_f32.c.o
[  2%] Building C object BasicMathFunctions/CMakeFiles/CMSISDSPBasicMath.dir/arm_dot_prod_f32.c.o
[  2%] Building C object BasicMathFunctions/CMakeFiles/CMSISDSPBasicMath.dir/arm_mult_f32.c.o
[  2%] Building C object BasicMathFunctions/CMakeFiles/CMSISDSPBasicMath.dir/arm_negate_f32.c.o
[  2%] Building C object BasicMathFunctions/CMakeFiles/CMSISDSPBasicMath.dir/arm_offset_f32.c.o
[  2%] Building C object BasicMathFunctions/CMakeFiles/CMSISDSPBasicMath.dir/arm_scale_f32.c.o
[  2%] Building C object BasicMathFunctions/CMakeFiles/CMSISDSPBasicMath.dir/arm_sub_f32.c.o
[  3%] Building C object BasicMathFunctions/CMakeFiles/CMSISDSPBasicMath.dir/arm_abs_f16.c.o
[  3%] Building C object BasicMathFunctions/CMakeFiles/CMSISDSPBasicMath.dir/arm_add_f16.c.o
[  3%] Building C object BasicMathFunctions/CMakeFiles/CMSISDSPBasicMath.dir/arm_clip_f16.c.o
[  3%] Building C object BasicMathFunctions/CMakeFiles/CMSISDSPBasicMath.dir/arm_dot_prod_f16.c.o
[  3%] Building C object BasicMathFunctions/CMakeFiles/CMSISDSPBasicMath.dir/arm_mult_f16.c.o
[  4%] Building C object BasicMathFunctions/CMakeFiles/CMSISDSPBasicMath.dir/arm_negate_f16.c.o
[  4%] Building C object BasicMathFunctions/CMakeFiles/CMSISDSPBasicMath.dir/arm_offset_f16.c.o
[  4%] Building C object BasicMathFunctions/CMakeFiles/CMSISDSPBasicMath.dir/arm_scale_f16.c.o
[  4%] Building C object BasicMathFunctions/CMakeFiles/CMSISDSPBasicMath.dir/arm_sub_f16.c.o
[  4%] Building C object BasicMathFunctions/CMakeFiles/CMSISDSPBasicMath.dir/arm_abs_q31.c.o
/tmp/ccd7xOuW.s: Assembler messages:
/tmp/ccd7xOuW.s:38: Error: unknown mnemonic `qsub' -- `qsub x4,x5,x4'
make[2]: *** [BasicMathFunctions/CMakeFiles/CMSISDSPBasicMath.dir/build.make:440: BasicMathFunctions/CMakeFiles/CMSISDSPBasicMath.dir/arm_abs_q31.c.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:350: BasicMathFunctions/CMakeFiles/CMSISDSPBasicMath.dir/all] Error 2
make: *** [Makefile:91: all] Error 2

christophe0606, 2022-06-29:
@bobkocisko So it looks like the intrinsics `qsub` is not available on your Cortex-A.
It is a bit weird. It is available on low end Cortex-A so I was assuming it would also be available on yours. Perhaps there is a compilation option. I don't know Cortex-A instruction set in detail so it is something I'll have to check.

The second warning with `vlogq_f32` is not a problem. I'll have to correct this but it does not prevent it from being built and used.

For the last problem : 
According to https://developer.arm.com/documentation/dui0491/e/C-and-C---Compiler-Implementation-Limits/Limits-for-integral-numbers , ULONG_MAX is ok and 32 bits.

But the value displayed by the error (18446744073709551615) is too big and on 64 bits as if the ULONG_MAX had been sign extended to 64 bits ... 

And according to this page : https://developer.arm.com/architectures/instruction-sets/intrinsics/ the `vdupq_n_u32` is using a 32 bits arguments.

So I don't really know what's happening here. But at least, the converted value is right so the code should still work.

christophe0606, 2022-06-29:
@bobkocisko I had forgotten a key point : I think you don't need to include `Core_A/Include` or `Core/Include` at all. You also don't need to define `ARM_MATH_DSP`.

Just define `-D__GNUC_PYTHON__` instead.

It is a flag I am using to build the Python wrapper and which is bypassing the Core includes. 

By defining this, you'll get the definition of the compiler macros used by CMSIS-DSP:

```C
#define  __ALIGNED(x) __attribute__((aligned(x)))
#define __STATIC_FORCEINLINE static inline __attribute__((always_inline)) 
#define __STATIC_INLINE static inline
```

and CMSIS-DSP will define a C version of the intrinsics not available on Cortex-A like `QSUB`.

So everything should build (including the fixed point arithmetic like q31, q15 and q7 versions).

bobkocisko, 2022-06-29:
Success!  Thanks so much @christophe0606!

Here is my (presumably last) commit to support our cpu: https://github.com/bobkocisko/CMSIS_5/commit/ec985095b83e1b88b0dec7f4afcac28d1287439b

christophe0606, 2022-06-30:
@bobkocisko With the `__GNUC_PYTHON__`  and `ARM_MATH_NEON` you still have all the Neon instructions.

It only disable instructions like `QSUB16`, `QADD8` etc ... which are useful for fixed point arithmetic algorithms (q7 and q15 format mainly) and only useful for the scalar parts. Vector part (when there is one) is using Neon.

The only effect of  `__GNUC_PYTHON__` in addition to defining some compiler macros, is to define a C implementation for those missing instructions. The C implementation is provided by `dsp/none.h` which means there is no DSP extensions on the core (DSP extension is this set of instruction `QADD8` etc ...).

On a big core, it should not make a lot of differences if you don't have those instructions and use the pure C implementation. Those instructions are useful on a Cortex-M.

I think those instructions are only available in aarch32 (but I don't know aarch64 very well).

christophe0606, 2022-06-30:
I am closing the github issue because we are moving CMSIS-DSP to a new repository : https://github.com/ARM-software/CMSIS-DSP

I am migrating the github issues to the new repository.

You can still comment on this closed issue or you can reopen a new one on the new repository if you have additional questions.

closed 2022-06-30 by christophe0606

bobkocisko, 2022-06-30:
Thanks again @christophe0606!