ARM-software/CMSIS_5 #1514: Latest KEIL Utilities ( 5.9.0) have unresolved dependencies

praedicow, 2022-06-23:
Hello,

currently we're using KEIL Utilities v 5.8.0 for our KEIL-Pack generation flow on a Linux server. I tried to update our flow to the latest KEIL Utilities (SVDConv and PackChk) to Version 5.9.0, but the problem occurred, that there are some dependencies which our system can not resolve.

This didn't occur in any of the older versions of these tools. So my question is, if it's possible, that you could switch back these tools to static linked executables?

Thank you for your support.
Alex

praedicow, 2022-06-23:
[image]

JonatanAntoni, 2022-06-23:
Hi @praedicow,

Thanks for pointing this out. This seems to be a mistake happened in SVDConv.

Do you have any issues with latest [PackChk 1.3.95](https://github.com/Open-CMSIS-Pack/devtools/releases/tag/tools%2Fpackchk%2F1.3.95) as well? PackChk is built with `-static-libc++` so only the `libc` is used dynamically. Is this sufficient or do we need to use `-static` instead to force full static linkage?

Cheers,
Jonatan

praedicow, 2022-06-23:
Hey Jonatan, 

thank you for your fast reply. We do not have the issue in _PackChk 1.3.95_, so dynamic linkage of `libc` is okay for us, but to reduce dependencies, full static linkage may be more robust.

The older versions of PackChk and SVDConv used full static linkage.

Best regards
Alex

JonatanAntoni, 2022-06-23:
Hi Alex,

Yes, that's true. Thanks for confirming that `libc` is okay for you. We need to discuss potential other side-effects.
But I think we will at least add `-static-libc++` for SVDConv.

Cheers,
Jonatan