ARM-software/CMSIS_5 #1515: L modifier in *_Msk macro-definitions of device header for 16bit registers

praedicow, 2022-06-23:
After generating the device header file by **SVDConv** with the `--fields=macro` option, all precompiler defines for `*_Msk ` have the modifier `L`, regardless of the actual register bitwidth.

Linting tools are reporting failures in the assignment of a _LONG_ value to a _SHORT_ register. 

It would be nice to adjust the `*_Msk` precompiler defines in the device header file, according to the actual register size in the _Pos/Mask Peripheral Section_.

[image]

JonatanAntoni, 2022-06-23:
Hi @praedicow,

Thanks for getting in touch. Can you please provide a reproducing SVD file?
I guess in code generation we blindly assume 32-bit registers whenever creating pos and mask defines.

Cheers,
Jonatan

JonatanAntoni, 2022-06-23:
Hi Alex,

Thanks for the example. The colleagues have already picked that up.

I wonder how a good and generic solution could look like. Unfortunately, C does not have integer suffixes for short (16-bit) and char (8-bit) length values. And the suffix `U` still relates to an `unsigned integer` which is 32-bit. 

The only solution I am aware of would be adding C-style casts so that the integer constants get casted to the register data type.

What do you think?

Cheers,
Jonatan

praedicow, 2022-06-24:
Hello, first of all you're completely right, skipping the `L` would also lead to 32bit values for ARM systems. We didn't think about that before. 
We discussed this yesterday and the only generic solution we came up with, was the one @thorstendb-ARM already presented. So yes, I think a specific typecast would  be a good solution. We weren't 100% sure if the `U` could still be necessary for Hex values, but most likely the compiler is aware of the unsigned datatype with the `uint` cast.

One additional comment, we think the casting is only necessary for the `*_Msk` defines, since the `*_Pos` will only be used for shifting operation. So it's up to you if you want to leave the `*_Pos` defines as they are or somehow match the casting format.

We're really happy with you're fast and good replies. Thank you.

JonatanAntoni, 2022-06-27:
@praedicow,

Thanks for you confirmation. We'll pick that up.

Cheers,
Jonatan

billybednar, 2023-08-16:
Defining constants using casts should generally be avoided because they're not valid in `#if` directives.

For registers smaller than `int`, adding casts seems unlikely to help in most cases. You can't really do anything with such a value without it being promoted to `int`. For example, the type of the expression `(short)12 & (short)6` is `int` as it actually means `(int)(short)12 & (int)(short)6`. If your tooling complains about assigning `long` to `short`, it will probably still complain about assigning `int` to `short` after adding casts.

I'd still suggest using the smallest suffix that's guaranteed to be wide enough: nothing/U for <= 16 bits, L/UL for 17 to 32, and LL/ULL for > 32

There's also the standard `[U]INTx_C` macros in stdint.h, which can be used to add an appropriate suffix. A possible downside is that they do follow the integer promotion rules, so `UINT8_C(123)` expands to `123` where `123U` might be preferable.

JonatanAntoni, 2023-09-05:
I'd like to point you to the new location where development of SVDConv happens in the open. See
https://github.com/Open-CMSIS-Pack/devtools/tree/main/tools/svdconv. May I ask you to re-raise your request there? Or even better: Contribute the change via pull-request. Thanks.