ARM-software/CMSIS_5 #1557: Array access out of bounds in __cmsis_start()

zhusbj, 2022-09-01:
https://github.com/ARM-software/CMSIS_5/blob/develop/CMSIS/Core/Include/cmsis_gcc.h
line 152 - 154

  for(uint32_t i=0u; i<pTable->wlen; ++i) {
      pTable->dest[i] = pTable->src[i];
    }

pTable->dest and  pTable->src are uint32_t types，but pTable->wlen is the array size in bytes. 
They are not matched and would casue array access out of bounds.

JonatanAntoni, 2022-09-01:
Hi @zhusbj, 

I think this is a misunderstanding of what the code is doing.

These for-loops are basically mem copies where `pTable->src` is pointing to the source, `pTable->dest` is pointing to the target, and `pTable->wlen` is giving the amount of 32-bit memory words to be copied.

The values of `pTable` are Linker generated and of course must match the memory layout. A mistake in the Linker script (which provides wrong values) may cause memory corruption or faults.

Does this make sense  to you?

Cheers,
Jonatan

zhusbj, 2022-09-02:
Hi Jonatan,

Sorry I referred to the older code CMSIS v5.6.0, where the .copy.table was defined as below. 
 .copy.table :
  {
    . = ALIGN(4);
    __copy_table_start__ = .;
    LONG (__etext)
    LONG (__data_start__)
    LONG (__data_end__ - __data_start__)
    __copy_table_end__ = .;
  } > FLASH

Now I checked the latest version and found it has been fixed by LONG ((__data_end__ - __data_start__) / 4).
It is no problem now. Thanks for you explanatio.

B.R.
Shaobo Zhu