ARM-software/CMSIS_5 #1572: In which case M33 generate osRtxErrorStackUnderflow but no " STKOF UsageFault "?

swordligit, 2022-10-26:
Hi, experts.
     I encountered an  osRtxErrorNotify which error was osRtxErrorStackUnderflow, and
then I dumped the thread info, the thread->sp is something like 0x20027ac8, and the
thread->stack_mem is 0x20027b00, and the osRtxThreadStackCheck found there
was a stack underflow.
    The case was generated in osRtxTick_Handler, which means current thread's
round robin timeout was used up, and will switch to another ready thread. before
scheduling out, stack checking failed.
    But that was very strange, because if the stack pointer is below the stack_limit, my understanding
is there should be a stack overflow exception. I did a test and allocated a big variable that larger
than stack, the thread did generate the exception.
    Another info is currently the PSP register is 0x20028098, PSPLIM is 0x20027b00,
that means the thread->sp was set at last schedule-out time.
    My cpu is cortex-m33, STAR-MC1, and the OS is RTX5.
Br,
Yingchun

RobertRostohar, 2022-10-26:
Hi,
please check if you are using the latest RTX V5.5.4. Stack overrun checking has been enhanced recently (V5.5.3).

swordligit, 2022-10-27:
Hi, Robert
Thank for your reply. you mean the commit:

commit 88f2b4fc1fb814dfc562b4555dc16640fd9537e2
Author: Robert Rostohar <Robert.Rostohar@arm.com>
Date:   Mon May 17 08:25:58 2021 +0200

    RTX5: enhance stack overrun checking

Yes, I am using 5.5.1 and don't have this enhancement.  I saw the
osRtxThreadStackCheck has been removed from osRtxThreadSwitch 
to the exactly context switch function SVC_ContextSaveSP.

But I still can not figure out why there is no stack overflow exception.
IMO there is only one case that the thread->sp was set abnormally (
but seems impossible,the sp is set in SVC_ContexSave3 just before schedule out last time).
so if the PSP was wrong, it should exception before.

RobertRostohar, 2022-10-28:
Could you try RTX 5.5.4 to see if the problem is gone?

swordligit, 2022-10-28:
OK,but this bug is not reproduced easily, so need time to debug this.
could you explain this patch?  the git log doesn't give so much details.

swordligit, 2022-11-19:
Still can not reproduce the problem, so close this issue currently, I would reopen
this issue if I can reproduce it again.
Thank you @RobertRostohar .

closed 2022-11-19 by swordligit