ARM-software/CMSIS_5 #1590: How to Describe Bitfields That Span Multiple Registers in CMSIS-SVD

bordeaux15, 2022-11-21:
I'm working on a project where bitfields can span multiple registers (i.e. 8-bit registers, 23-bit bitfields across 3 registers). Example below:

> Bitfield_A is 23 bits across 3 8-bit registers Register_1, Register_2, Register_3
> Register_1 contains Bitfield_A[7:0]
> Register_2 contains Bitfield_A[15:8]
> Register_3 contains Bitfield_A[22:16]

I don't see any where in the CMSIS-SVD 1.3.9 schema that can support this structure. The bitRange identifier is used only to specify a bitfields range within a register. But it cannot specify which bits of the bitfields are included in that bitRange. A couple questions:

- Is there a method to support these bitfields split across multiple registers that I'm not seeing?
- Is it possible to add a schema/tag into the bitfield definition that would highlight which bits of given bitfield occur in a given register?

thorstendb-ARM, 2022-11-21:
Hi, does the register have a 32bit representation (means, all three (four) can be read as one 32bit register having the upper 8 bit "reserved")? In this case you can use both descriptions (1x 32bit, 3x 8bit) as alternate specification ("union concept").

JonatanAntoni, 2022-11-21:
Please note that CMSIS-SVD spec has moved to https://github.com/Open-CMSIS-Pack/svd-spec.

bordeaux15, 2023-02-07:
Hi @thorstendb-ARM,

The register doesn't have a 32 bit representation only the multiple 8 bit representations. I need to know the schema used for naming the LSB register starts with _1 and up to _3.

JonatanAntoni, 2023-02-08:
Hi @bordeaux15,

I suggest to move this discussion to https://github.com/Open-CMSIS-Pack/svd-spec.
Perhaps raise another issue over there and describe the layout you want to achieve in C code for instance. Do you want to have access to the bitfield with a single read/write access? Or do you need three separate 8-bit accesses? Please take care that LSB/MSB definition depends on hardware representation.

Cheers,
Jonatan

thorstendb-ARM, 2023-02-21:
Can you please give an example how you would describe this in C with struct and union?