ARM-software/CMSIS_5 #1598: how to know remote version from a specific commit id

brainchip-india, 2022-12-17:
Hi, I am downloading tensorflow lite of a specific commit ID([@713b6](https://github.com/tensorflow/tflite-micro/tree/713b6ed6bd81d8d6906d885e14f444aaf9c154f6) in that from the source [ext_libs](https://github.com/tensorflow/tflite-micro/tree/713b6ed6bd81d8d6906d885e14f444aaf9c154f6/tensorflow/lite/micro/tools/make/ext_libs)/cmsis_download.sh) it downloading a specific commit id of CMSIS repo.

My question is how to find a CMSIS remote version from that commit ID?

Thanks in advance.

felix-johnny, 2022-12-19:
@brainchip-india Thanks for the question. There are two cases to consider. 

1. When CMSIS-NN was part of CMSIS_5 repository(Prior to TFLM PR https://github.com/tensorflow/tflite-micro/pull/1468)
The commit ID that you find in cmsis_download.sh is the commit ID of CMSIS_5 repository as well. i.e, they are one and the same.

2. Current state where CMSIS-NN is its own repository. 
The commit ID that you find in cmsis_download.sh is the commit ID of CMSIS_5 repository.  The commit ID that you find in cmsis**_nn_**download.sh is the commit ID of CMSIS-NN repository.

brainchip-india, 2022-12-22:
Thanks @felix-johnny

closed 2022-12-22 by brainchip-india