ARM-software/CMSIS_5 #1604: Inconsistent Naming Scheme In core_cmXX.h Files

Devilbinder, 2023-01-26:
Please see the following 2 line. The following two macros do not have the same capitalization. Is this intentional?

Found this when porting a code base from a cm4 to a cm33 as one does with this chip shortage. 

https://github.com/ARM-software/CMSIS_5/blob/2e02216ab271717e5c7512c88a1ec54275940bd5/CMSIS/Core/Include/core_cm33.h#L1067

https://github.com/ARM-software/CMSIS_5/blob/2e02216ab271717e5c7512c88a1ec54275940bd5/CMSIS/Core/Include/core_cm4.h#L857

jkrech, 2023-01-27:
Probably we require to duplicate one MACRO with different capitalization to maintain backward compatibility.

JonatanAntoni, 2023-01-27:
Searching for `ITM_TCR_TraceBusID_Pos` points out this inconsistency affects Cortex-M3, -M4, and -M7. And not only TraceBusID, but other macros as well. I think (except the suffixes `_Pos` and `_Msk`) we should agree on a common capitalization in macros. Typically, macros with constants are all uppercase.

GuentherMartin, 2023-01-27:
The 'older' macros use capital names and the 'newer' ones use the same names as in the manuals.