ARM-software/CMSIS_5 #1605: Propose add CPUECTLR access functions.

ua1arn, 2023-02-11:
How about adding this functions to _CMSIS/Core_A/Include/cmsis_cp15.h_ ?

`
#if (__CORTEX_A == 8U)

#define CPUECTLR_SMPEN_Msk (1u << 6)	// SMPEN 1: Enables data coherency with other cores in the cluster.

// 4.5.77 CPU Extended Control Register

/** \brief  Get CPUECTLR
    \return               CPU Extended Control Register, EL1
 */
__STATIC_FORCEINLINE uint64_t __get_CPUECTLR(void)
{
	uint64_t result;
  __get_CP64(15, 1, result, 15);
  return(result);
}

/** \brief  Set CPUECTLR
    \param [in]    cpuectlr  CPU Extended Control Register, EL1
 */
__STATIC_FORCEINLINE void __set_CPUECTLR(uint64_t cpuectlr)
{
	__set_CP64(15, 1, cpuectlr, 15);
}

#endif /* (__CORTEX_A == 8U)  */
`

ua1arn, 2023-02-12:
Also, essential addition is CPUACTRLR with codes 15. 0, xx, 15.

GuentherMartin, 2023-02-16:
Hello @ua1arn, are these changes required to support Cortex-A8 devices? You mention a dedicated chapter `4.5.77 CPU Extended Control Register`.  Could you point me to the document where I can find this information?

ua1arn, 2023-02-17:
You right. CPUECTLR bs a Cortex-A53 specific register (like different-specs for Cortex-A7 and Cprtex-A9 for ACTLR.
Mentioned document is `DDI0500J_cortex_a53_r0p4_trm.pdf`
All these updates tested on Allwinner A64.
May be right paramerer are:
```
#define __CORTEX_A                   53U      /*!< Cortex-A# Core                              */
#define __CA_REV                 0x0000U      /*!< Core revision r0p0                          */
#define __FPU_PRESENT                 1U      /*!< Set to 1 if FPU is present                  */
#define __GIC_PRESENT                 1U      /*!< Set to 1 if GIC is present                  */
#define __TIM_PRESENT                 1U      /*!< Set to 1 if TIM is present                  */
#define __L2C_PRESENT                 0U      /*!< Set to 1 if L2C is present                  */

```

ua1arn, 2023-02-17:
For reference, predefined sybols fo gcc wirg -mcpu=cortex-a53 in attachments.
[defines_aarch64.txt](https://github.com/ARM-software/CMSIS_5/files/10762362/defines_aarch64.txt)
[defines_aarch32.txt](https://github.com/ARM-software/CMSIS_5/files/10762363/defines_aarch32.txt)

GuentherMartin, 2023-02-20:
Hello @ua1arn,
if CPUECTLR is a Cortex-A53 specific register then I would suggest to guard the functions with macro `#if (__CORTEX_A == 53U)`. Macro `#define __CORTEX_A  53U` needs to be configured in the device specific header file.
Please let me know you opinion.