ARM-software/CMSIS_5 #1606: CMSIS-DSP: arm_sin_cos_example_f32.c, How to disable unused functions?

Rimatara, 2023-02-12:
I try to compile [arm_sin_cos_example_f32.c](https://arm-software.github.io/CMSIS_5/DSP/html/arm_sin_cos_example_f32_8c-example.html#a12) in the MPLAB X IDE, XC32, Harmony 3 but I get an error:

**commontables.o: Link Error: Could not allocate section .rodata, size = 920376 bytes, attributes = code 
 Link Error: Could not allocate program memory**

How and where I can disable unused functions to fit program to ATSAMD21J18 (Cortex-M0+), Data/Program 32k/256k?

[Sin_Cos_Example_f32.zip](https://github.com/ARM-software/CMSIS_5/files/10716945/Sin_Cos_Example_f32.zip)

JonatanAntoni, 2023-02-13:
Hi @Rimatara,

please take note that CMSIS-DSP has move off into its own repository at https://github.com/ARM-software/CMSIS-DSP. You might want to raise your question over their.

Cheers,
Jonatan

Rimatara, 2023-02-13:
Hi Jonatan,

Thank you for letting me know. I posted it there.

Have a nice day,
Dan