ARM-software/CMSIS_5 #1612: Inverted parameters in MMU_GetSectionDescriptor() and MMU_GetPageDescriptor()

ANOSTM, 2023-03-22:
Hello,

In CMSIS v5.8.0 (and also in v5.9.0), some parameters are inverted in CMSIS/Core_A/Include/core_ca.h
The call of MMU_APPage and MMU_APSection are not aligned with their definitions ( user <> priv).
See below

- MMU_GetSectionDescriptor()
[image]

- MMU_GetPageDescriptor()
[image]

JonatanAntoni, 2023-03-22:
Thanks for pointing this out.
May I ask you to raise this change as a PR against develop branch, please?

ANOSTM, 2023-03-22:
Could you confirm that you mean Pull Request by PR ?

JonatanAntoni, 2023-03-22:
Yes, I mean Pull Request ;)

ANOSTM, 2023-03-22:
I created a pull request from a fork : https://github.com/ANOSTM/CMSIS_5/pulls
Is it OK for you ?

JonatanAntoni, 2023-04-13:
You need to raise the PR against this repo, not internally to your personal fork.

ANOSTM, 2023-04-13:
Hello,

After some discussion, I raised https://github.com/ARM-software/CMSIS_5/pull/1613
I saw this PR (1613) has been merged.
So probably the patch is already available, and this PR (1612) can be closed.
Could you confirm ?

GuentherMartin, 2023-04-13:
Hello @ANOSTM, this PR can be closed.
The requested changes are already merged with #1613

JonatanAntoni, 2023-04-13:
Thanks, @GuentherMartin, I overlooked that.

closed 2023-04-13 by JonatanAntoni