ARM-software/CMSIS_5 #1618: Main Cortex-M55/M85 EWIC registers not present

kjbracey, 2023-04-14:
core_cm55.h and core_cm85.h have `EWIC` and `EWIC_Type` definitions, but these describe the "EWIC interrupt status access" registers in the core that provide access to the NVIC for the benefit of transferring state to and from the EWIC.

There are no definitions for the EWIC itself at E0047000, which software is more likely to want to access - at a minimum you need to enable it in its control register.

GuentherMartin, 2023-04-17:
Hello @kjbracey, thank you for pointing out.
I did not pay enough attention to the usage when adding these registers.

My proposal is to change the current EWIC type and defines from `EWIC_Type` to `EWIC_ISA_Type` (EWIC Interrupt status access) and add the EWIC register under  `EWIC_Type` (see chapter  A.2 EWIC register summary in Cortex-M55 TRM).

Please let me know you opinion.

GuentherMartin, 2023-04-17:
Hello @kjbracey, please check https://github.com/ARM-software/CMSIS_5/pull/1620/commits/e04fcc412187016647c99cc0c1cc55c35295420a.

I was not able to add you as reviewer. :-(

kjbracey, 2023-04-17:
Left some comments on the first version of the commit.