ARM-software/CMSIS_5 #1635: Issue with osDelay in TC_osKernelResume_1

ianfield, 2023-06-19:
cmsis_rtos2_validation/Source/RV2_Kernel.c

Entry to TC_osKernelResume_1 (comes after TC_osKernelSuspend_1) the kernel is in state osKernelSuspended.

What is the expected return value for osDelay(1U). Should it be:

  ASSERT_TRUE(osDelay(1U) == osError);

The comment "/* Synchronize tick */" seems out of place if the system timer is disabled at that point.

ianfield, 2023-06-19:
Alternatively, if kernel is suspended on entry, would this make sense:

    osKernelResume(0);

    ASSERT_TRUE(osDelay(1U) == osOK);

VladimirUmek, 2023-06-20:
Hi,
this should be open in CMSIS-RTOS2_Validation, but I guess we can also solve it here.

If you are referring to [TC_osKernelSuspend_1](https://github.com/ARM-software/CMSIS-RTOS2_Validation/blob/ca0a96eda93297252adfe376d61f112ec90acfe5/Source/RV2_Kernel.c#L402) and [TC_osKernelResume_1](https://github.com/ARM-software/CMSIS-RTOS2_Validation/blob/ca0a96eda93297252adfe376d61f112ec90acfe5/Source/RV2_Kernel.c#L480) then [osDelay(1U)](https://github.com/ARM-software/CMSIS-RTOS2_Validation/blob/ca0a96eda93297252adfe376d61f112ec90acfe5/Source/RV2_Kernel.c#L488) is there to set the tick counter to the start of time slice (as close as possible, hence it is synchronizing tick). It sets the initial condition to ensure correct test case flow.

Each test is independent and looking at the code, the kernel shall be running after [TC_osKernelSuspend_1](https://github.com/ARM-software/CMSIS-RTOS2_Validation/blob/ca0a96eda93297252adfe376d61f112ec90acfe5/Source/RV2_Kernel.c#L402). The kernel is [resumed](https://github.com/ARM-software/CMSIS-RTOS2_Validation/blob/ca0a96eda93297252adfe376d61f112ec90acfe5/Source/RV2_Kernel.c#L431) and since osKernelSuspend and osKernelResume shall not be called from ISR any other call shall not suspend the kernel. [TC_osKernelResume_1](https://github.com/ARM-software/CMSIS-RTOS2_Validation/blob/ca0a96eda93297252adfe376d61f112ec90acfe5/Source/RV2_Kernel.c#L480) is therefore called with the kernel tick running.

Does this make sense?