ARM-software/CMSIS_5 #1636: RTX4 EXCLUSIVE_ACCESS macro undefined for GCC compiler

tobermory, 2023-06-20:
I am aware that RTX4 code is several years old and no longer supported. I do wonder if any code author could comment on rt_HAL_CM.h's undef of the __USE_EXCLUSIVE_ACCESS macro when a GNU compiler is used:

```
#elif defined (__GNUC__)        /* GNU Compiler */
#undef  __USE_EXCLUSIVE_ACCESS
#endif
#ifdef __USE_EXCLUSIVE_ACCESS
 #define rt_inc(p)     while(__strex((__ldrex(p)+1U),p))
 #define rt_dec(p)     while(__strex((__ldrex(p)-1U),p))
#else
 #define rt_inc(p)     __disable_irq();(*p)++;__enable_irq();
 #define rt_dec(p)     __disable_irq();(*p)--;__enable_irq();
#endif
```
Was there some property of the GNU compiler toolchain at the time (2016?) such that the ldrex and strex instructions could not be used? 

I see in RTX5, namely `rtx_core_cm.h`, exclusive access IS available when the GNU compiler is used. In fact, toolchain checking is not employed, but rather some 'arch' tests:
```
#ifndef EXCLUSIVE_ACCESS
#if    ((defined(__ARM_ARCH_7M__)      && (__ARM_ARCH_7M__      != 0)) || \
        (defined(__ARM_ARCH_7EM__)     && (__ARM_ARCH_7EM__     != 0)) || \
        (defined(__ARM_ARCH_8M_BASE__) && (__ARM_ARCH_8M_BASE__ != 0)) || \
        (defined(__ARM_ARCH_8M_MAIN__) && (__ARM_ARCH_8M_MAIN__ != 0)))
#define EXCLUSIVE_ACCESS        1
#else
#define EXCLUSIVE_ACCESS        0
#endif
```

RobertRostohar, 2023-06-21:
"__ldrex" and "__strex" were Arm Compiler 5 Intrinsics and later deprecated (5.06). They are not available for GCC.
Since there was no better handling for exclusive access in RTX4 at that point, it was disabled for GCC.
RTX5 uses assembly for exclusive operations available for Arm Compiler 5/6, GCC and IAR.

tobermory, 2023-06-21:
Thank you, that makes perfect sense. I was hunting high and low through header files looking for those __ldrex, __strex calls.  Now I see why I never found them!