ARM-software/CMSIS_5 #1639: Add HOST support

higaski, 2023-07-26:
Any chance we could get HOST support one day like the [CMSIS-DSP](https://github.com/ARM-software/CMSIS-DSP)? Basically what I'm asking for are `cmsis_*.h` files without any inline assembly. Currently I'm maintaining one file for GCC myself and it's really cumbersome.

And while we're at it, we might as well add CMake support. :wink:

JonatanAntoni, 2023-07-26:
Thanks for your request. May I ask you to jump over to [ARM-software/CMSIS_6](https://github.com/ARM-software/CMSIS_6), please?

We are always happy to receive contributions. CMSIS-Core abstracts processor core peripherals which are not available on amd/arm host systems. Please feel free to write down some requirements into an issue. Giving some usage examples or even projects would be helpful to understand the request.

CMake is already supported via CMSIS-Build, see [Open-CMSIS-Pack/cmsis-toolbox](https://github.com/Open-CMSIS-Pack/cmsis-toolbox).

higaski, 2023-07-26:
Thanks for the fast reply, I've moved my request to CMSIS_6:
https://github.com/ARM-software/CMSIS_6/issues/32