ARM-software/CMSIS_5 #1648: EMAC_iMXRT1064.c file is not compatible with fsl_enet driver in Keil

DanielMtzNXP, 2023-10-10:
EMAC_iMXRT1064.c file included from Ethernet MAC (API) in CMSIS Driver on Manage Run-Time Environment window, is no longer compatible with fsl_enet driver version ,2.7.0. this new version is available on http://www.keil.com/pack/Keil.iMXRT1064_MWP.1.5.1.pack

EMAC_iMXRT1064.c file has the ENET_SetCallback function that was removed from this fsl_enet driver revision and the EMAC_iMXRT1064.c file still has this function this is causing a compilation error when the EMAC_iMXRT1064.c file is included on an ethernet demo.

RobertRostohar, 2023-10-11:
Hi @DanielMtzNXP,

The EMAC driver for iMXRT1064 (EMAC_iMXRT1064.c) in Keil.iMXRT1064_MWP_V1.5.1 pack is based on the fsl_enet driver 2.5.4 from NXP.MIMXRT1064_DFP.15.1.0 pack (SDK package V2.12.1).

This EMAC driver is no longer compatible with the latest fsl_enet driver, as you have already identified.

The updated [Keil.iMXRT1064_MWP.1.6.0 pack](https://www.keil.com/pack/Keil.iMXRT1064_MWP.1.6.0.pack) which has just been released, includes also the updated EMAC driver based on the fsl_enet driver 2.7.0 from NXP.MIMXRT1064_DFP.17.0.0 pack (SDK package V2.14.0).

JonatanAntoni, 2023-10-11:
Gents, If I am reading this correctly, the breaking change got introduced into fsl_enet driver between versions 2.5.4 and 2.7.0. According to semantic versioning, incrementing the minor part of the version number suggests backward compatibility, i.e., only new features added.

I highly recommend to take semantic versioning serious as such issues can have nasty and cumbersome impact to our uses application projects. As integrators, our users want to combine software components from lots of different vendors. This can become a nightmare outweihgting the benefits of reusable software components and CMSIS-Packs.