ARM-software/CMSIS_5 #1649: `writeConstraint` in `register` crushes SVDConv

kossnikita, 2023-10-27:
To reproduce

- Add `writeConstraint` with any content to `register` (not `field`)
- run `SVDConv <svd>`
- exit code -1073741819 with no error message

Windows 11 64bit

```
CMSIS-SVD SVD Consistency Checker / Header File Generator V3.3.42
```

kossnikita, 2023-10-27:
Looks like it already fixed in [Open-CMSIS-Pack/devtools](https://github.com/Open-CMSIS-Pack/devtools/releases/tag/tools%2Fsvdconv%2F3.3.46)

jkrech, 2023-10-27:
Please note we are getting close to releasing CMSIS_6
This means there will not be another pack release from this repo.
Also note that SVDConv will not be shipped as of pack going forward.
As you rightly state above the SVDConv has been open sourced and is available as part of the Linaro Open-CMSIS-Pack github organization: https://github.com/Open-CMSIS-Pack
Please note that SVDConv pre-build binaries are now distributed as part of the CMSIS-Toolbox:
For more information see: https://github.com/Open-CMSIS-Pack/cmsis-toolbox/releases
Please raise any issues seen with any tools contained in the CMSIS-Toolbox here:
https://github.com/Open-CMSIS-Pack/cmsis-toolbox/issues/new/choose

I will close the issue here.

closed 2023-10-27 by jkrech