ARM-software/CMSIS_5 #1651: Thread switch on osKernelRestoreLock?

juliobc, 2023-11-08:
Hi,

I'd like to know if at the time I execute an osKernelRestoreLock() then a Thread switching is executed.

If a Thread with higher priority goes to a ready state while the kernel is locked, is there a thread switching when osKernelRestoreLock() is executed? Or the thread switching is executed in the next thread switching x msecs after?

Regards

JonatanAntoni, 2023-11-09:
Hi @juliobc,

Every service call in RTX5 is a scheduling point.
So, yes, once the kernel is resumed a thread switch shall happen.
If you call RestoreLock from within interrupt context, the thread switch is deferred to the PendSV running on lowest exception priority.

Is this what you are looking for?

juliobc, 2023-11-09:
Hi @JonatanAntoni 

Great. This is what I was looking for.
thank you very much