ARM-software/CMSIS_5 #1666: Git LFS Data Limit Exceeded

jonas-eschmann, 2024-05-01:
Hi,

I'm getting this error when initializing CMSIS (as a submodule from another repo):

```
Error downloading object: CMSIS/DSP/Lib/ARM/arm_ARMv8MBLl_math.lib (496f8cd): Smudge error: Error downloading CMSIS/DSP/Lib/ARM/arm_ARMv8MBLl_math.lib (496f8cdab57174e66eb6cfb365cba46298b86440ddaf1660176bcac0566aebb6): batch response: This repository is over its data quota. Account responsible for LFS bandwidth should purchase more data packs to restore access.
```
It looks like this is a problem with the Github LFS hosting data quota of your organization. Would be great if you could make this available again.

Best regards!
Jonas

JonatanAntoni, 2024-05-02:
Hi @jonas-eschmann,

I expected this will cause issues at some point. Sorry to hear that.

Do you have any specific reasons for cloning the CMSIS repo as submodule?
Can you either rely on the latest CMSIS 5 release pack. Or if possible update to CMSIS 6 et al and get them from their new locations?

Cheers,
Jonatan

JonatanAntoni, 2024-05-02:
Duplicates #1201

jonas-eschmann, 2024-05-02:
Sorry for the duplicate, I dindn't look that fare back because I assumed it would have been a recent issue.

I'm using it because it is part of the Crazyflie firmware. The submodule is in:

https://github.com/bitcraze/crazyflie-firmware/tree/d9ee34a78cecf2eaf8b05e0b54487cff0c7b4590/vendor

It is really unfortunate that GH is so restrictive with the LFS quota but I guess it is downstream of the fact that cloud egress costs are high in general. Maybe it's time to finally implement http status 402 to solve this tragedy of the commons problem with microtransactions 😬

JonatanAntoni, 2024-05-06:
You might want to ask the maintainer of this project to rely on released packs instead of pulling from repositories directly.

closed 2024-05-06 by JonatanAntoni