ARM-software/CMSIS_5 #1670: Error[Li005]: no definition for "arm_cfft_init_1024_f32" [referenced from ..\IAR-CMSIS-DSP\Examples\cortexM4lf\Obj\arm_fft_bin_example\arm_fft_bin_example_f32.o]

vipinthomas123, 2024-07-18:
Apologies for a simple question here, I've followed the usage instructions and explored all option to resolve the issue but with no luck.

We are are having trouble in compiling the arm_fft_bin_example.ewp. There is linker error for the arm_cfft_init_1024_f32 function.

I have added below library - 
C:\Program Files\IAR Systems\Embedded Workbench 9.2\arm\CMSIS\Lib\IAR\iar_cortexM4lf_math.a

I have added below path for header files -
..\IAR-CMSIS-DSP\CMSIS_5\CMSIS\Core\Include
..\IAR-CMSIS-DSP\CMSIS-DSP\Include
..\IAR-CMSIS-DSP\CMSIS-DSP\Include\dsp

Development environment 
[image]

Are we missing any additional settings.

Additionally can anyone please confirm example is for 1024 samples but testInput_f32_10khz (test input data) is of 2048, it is always expected the data samples to be double of processing samples??

JonatanAntoni, 2024-07-18:
Hi @vipinthomas123,

Can you share the specific linker error you are getting?
For the details, I am pretty sure @christophe0606 can provide answers.

Please take note that we splittet off CMSIS-DSP into its own repo a while ago. You find it's new home at https://github.com/ARM-software/CMSIS-DSP. The library became its own pack, so it might be much easier to update it independently from other software components, now.

christophe0606, 2024-07-18:
@vipinthomas123 Note that the function `arm_cfft_init_1024_f32` is recent. If you're using a too old CMSIS-DSP you won't get it. Please open a new github issue on the CMSIS-DSP repository

vipinthomas123, 2024-07-19:
@JonatanAntoni Thanks for the information. 

@christophe0606 Thanks for the reply. will create the new issue on the CMSIS-DSP repository,

Closing this issue. Thank you.

closed 2024-07-19 by vipinthomas123