ARM-software/CMSIS_5 #1681: RTE_Components.h not defined CMSIS_device_header

julio-bc, 2024-12-28:
Hi,

I updated to MDK_5.41 with STM32F4 and it is not generating CMSIS_device_header in the file RTE_Components.h not

With MDK 5.40 I had:

[image]

And in MDK 5.41 I have:

[image]

and I have error in compilation:

[image]

I googled it and can't find the problem. Any idea?
Thank you

julio-bc, 2024-12-29:
Hi, I have:

[image]

if I select Keil::STM32F4xx_DFP to 2.17.1, the problem is fixed and CMSIS_device_header is defined in RTE_Components.h, but with 3.0.0 CMSIS_device_header is undefined

[image]

RobertRostohar, 2025-01-06:
The [STM32F4xx_DFP 3.0.0](https://github.com/Open-CMSIS-Pack/STM32F4xx_DFP/releases/tag/v3.0.0) pack has been updated for the new CMSIS-Toolbox CubeMX integration.

It is mandatory to use CubeMX, see more info about [usage](https://github.com/Open-CMSIS-Pack/STM32F4xx_DFP?tab=readme-ov-file#usage).

When `Device:CubeMX` component is selected the device header and the `CMSIS_device_header` define are available again.

closed 2025-01-14 by jkrech