ARM-software/CMSIS_5 #1684: Error inTPI_FFCR_EnFmt_Msk define in core_cm55.h

alexian79, 2025-05-26:
[image]

alexian79, 2025-05-26:
Aditionally I would like to ask regarding TPI_BASE, it is defined as 
#define TPI_BASE            (0xE0040000UL)                             /*!< TPI Base Address */
But for implementation in STM32n6 it should be changed to 
#define TPI_BASE            (0xE0084000UL) 
And I dont know is it ST moved it or it is also a typo?

JonatanAntoni, 2025-06-02:
> Aditionally I would like to ask regarding TPI_BASE, in the same file, it is defined as #define TPI_BASE (0xE0040000UL) /*!< TPI Base Address */ But for implementation in STM32n6 it should be changed to #define TPI_BASE (0xE0084000UL) And I dont know is it ST moved it or it is also a typo?

The [Reference Manual for STM32N6 series](https://www.st.com/resource/en/reference_manual/rm0486-stm32n647657xx-armbased-32bit-mcus-stmicroelectronics.pdf) lists two TPIU components, one in the processor rom table with address `0xE0040000` (but marked as not present) and another one in the trace subsystem rom table with address `0xE0084000`. I.e., the one in the CoreSight system is not the one exposed via the Cortex-M55 processor header.

alexian79, 2025-06-02:
Thank you, got it.